makerbase-mks/MKS-SGEN_L-V2 #1: Board does not work with UART

Kokorone, 2020-08-07:
I have the SGen L V2 and I cannot get UART mode to work with the mks tmc2209 drivers. I have added the jumpers in the correct configuration for uart under the driver as well as configured marlin with uart and m122 enabled. Running m122 returns a bad response on all the drivers, meaning they are not connected with uart.

[image]

Kokorone, 2020-08-08:
This was no longer a problem when connected to a proper power supply, I was having this issue exclusively through usb. Closing ticket

closed 2020-08-08 by Kokorone

makerbase-mks, 2020-08-08:
Can you provide your motherboard jumpers setting picture and your configuration.h file?

makerbase-mks, 2020-08-08:
OK!