makerbase-mks/MKS-SGEN_L-V2 #2: FAN LAYER

patxinet, 2020-08-14:
Hello, where can I connect the layer fan, so that I can regulate it with pwm, through the simplify, slicer, cura programs, etc.
do you have to modify, in marlin the outputs of the plate? is it fan1 or fan2?

#define BOARD_MKS_SGEN 2500 // MKS-SGen (Power outputs: Hotend0, Hotend1, Bed, Fan)
Thank you

makerbase-mks, 2020-08-15:
Hi! You can download MKS SGEN_L V2 path marlin firmware and it is support SGEN_L V2 motherboard. But offical marlin firmware, not support SGEN_L V2 cuerently. MKS has pull requests https://github.com/MarlinFirmware/Marlin/pull/18822
Usually, the motherboard FAN1 and FAN2 can be control by cammand,
M106 P0 S255  FAN1 OPEN
M106 P1 S255  FAN2 OPEN

makerbase-mks, 2020-09-23:
If there is no response, then I will close this question.