makerbase-mks/MKS-SGEN_L-V2 #6: Marlin 2.0

Kyosuke33, 2020-08-27:
Hello, i download the latest Marlin 2.0 but there is no Board MKS Sgen L V2.
You know when it will be availiable ?
Don't want to use yours but a fresh build.
Thank you.

savenko-egor, 2020-08-27:
You need copy [this](https://github.com/makerbase-mks/MKS-SGEN_L-V2/blob/master/Firmware/Marlin-2.0.6/Marlin/src/pins/lpc1769/pins_MKS_SGEN_L_V2.h) file to your Marlin folder.

Kyosuke33, 2020-08-27:
Ok, thank you !

makerbase-mks, 2020-08-31:
We are following up and have suggested that marlin first add support for the MKS SGEN_L V2 motherboard, which is in progress

makerbase-mks, 2020-09-02:
@Kyosuke33 @savenko-egor 
[image]
Last version marlin has add it.

closed 2020-09-02 by makerbase-mks