makerbase-mks/MKS-SGEN_L-V2 #7: How to wire BTT TFT35 v3.0 to MKS SGEN L v2?

ZhuDaHai, 2020-09-10:
The BigTree Tech TFT35 v3.0 TFT connector has 5v GND RX, TX and RST (reset) pins, but the MKS SGEN L v2.0 board AUX-1 connector does not have RST (reset).
How to hook up that display to this mainboard?
What needs to change in Marlin?
Thanks!

makerbase-mks, 2020-09-10:
MKS SGEN_L V1 or V2
```
AUX-1                                      
5V  GND  TX   RX
5V  GND  NC  NC

5V connect to 5V
GND connect to GND
TX connect to TFT's RX
RX connect to TFT's TX
RST connect to TFT's RST
```
[image]

ZhuDaHai, 2020-09-10:
It's labeled 'TRST' on the board's pin diagram. So that's the same as just 'RST' ?
Thanks!

makerbase-mks, 2020-09-10:
Sorry! You need connect RST to MKS SGEN_L V2 WIFI interface RST
[image]

ZhuDaHai, 2020-09-10:
Ok. Thanks!

closed 2020-09-10 by ZhuDaHai

jorgedelacasa, 2021-02-22:
Hola, buenas. Acabo de entrar y tengo dudas.Pro favor  al equipo de makerbase-mks.
Ayuda. He dejado de usar mi mkstft35 porque no logro comunicarme con mks sgen-l v1.0 y esta btt tft35 no se como ponerla.
Por favor ayudenme. Muchas gracias

SemStal, 2021-09-30:
I've got the same question but with the v 1.0 which doesn't have a wifi break out. Is there a reset pin I can pull from somewhere else? I believe the pins layout says it should be pin 16.