makerbase-mks/MKS-SGEN_L-V2 #8: firm ware useless it will not compile

helgjohn, 2020-09-10:
the files on mks site does not compile and if you use marlin original 2.x under lpc 1769 Mks sgenl ver 2 does not exist and if you use #define BOARD_MKS_SGEN  it compiles but noting will work no move ment and all cine of errors pop up. Whats going on here? 
There are some ready bin files but they are useless to me nothing is correct in bed size and the stuff i want enabled.

Hannesm90, 2020-09-10:
download the latest stable marlin software ( https://github.com/MarlinFirmware/Marlin )and only replace the buildroot folder, should fix your problem.

helgjohn, 2020-09-10:
> download the latest stable marlin software ( https://github.com/MarlinFirmware/Marlin )and only replace the buildroot folder, should fix your problem.
TANKS BUT NO LUCK 
YES IT SEEM TO WORK BUT NO COMPILE YET AS ALWAYS NEW PROBLEM 
( #error "TMC2208 or TMC2209 on Z2 requires Z2_HARDWARE_SERIAL or Z2_SERIAL_(RX|TX)_PIN.") 
My training wheels are a bit too small i am so lost in this so now i am relay lost and yelling HELP!!!

savenko-egor, 2020-09-11:
You need download [bugfix branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) from Marlin. PR approved only in bugfix branch.

Doyle4, 2020-09-11:
> You need download [bugfix branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) from Marlin. PR approved only in bugfix branch.
> For example pins file for SGEN L V2 in bugfix branch: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/lpc1769/pins_MKS_SGEN_L_V2.h

I have downloaded the Bugfix, but it shows it was created few months back, 2.0.6.1 is the latest.. how do I go about making bugfix 2.0.6.1 ?

Doyle4, 2020-09-11:
bugfix still shows unknown board MKS SGEN L V2

savenko-egor, 2020-09-11:
I think you need download stable (2.0.6.1) Marlin and change file from this PR: https://github.com/MarlinFirmware/Marlin/pull/18822

savenko-egor, 2020-09-11:
> bugfix still shows unknown board MKS SGEN L V2

I think you downloaded not bufix. Link on bugfix archive: https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip

Doyle4, 2020-09-11:
Sorry im a little confused as my old MKS GEN L V1 has had no problems.

Im unsure what PR is, downloaded bugfix, unknownboard, downloaded 2.0.6.1 and still unknown board

savenko-egor, 2020-09-11:
Okay =) No problem.
Download this archive: https://github.com/MarlinFirmware/Marlin/archive/2.0.6.1.zip
Unpack archive.
Download and replace this files:
https://raw.githubusercontent.com/mks-viva/Marlin/4bdaec14dbb68006ef9067307143d45ea0080996/Marlin/src/pins/pins.h
https://raw.githubusercontent.com/mks-viva/Marlin/4bdaec14dbb68006ef9067307143d45ea0080996/Marlin/src/core/boards.h
https://raw.githubusercontent.com/mks-viva/Marlin/4bdaec14dbb68006ef9067307143d45ea0080996/Marlin/src/pins/lpc1769/pins_MKS_SGEN_L_V2.h
Path for replace you can see in URLs:
Marlin/src/pins/pins.h
Marlin/src/core/boards.h
Marlin/src/pins/lpc1769/pins_MKS_SGEN_L_V2.h
Read this Wiki: https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/Marlin_firmware#configure-and-compile and change configuration files.
Done. Try compile your firmware.

If I try this, I got error:
```
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:00.810
```
But I think it problem in my vscode or python version.

Doyle4, 2020-09-11:
Thank you for your time, sadly at work now but will teat this when home! Thank you again.

helgjohn, 2020-09-11:
================================================= [SUCCESS] Took 86.06 seconds =================================================

Environment    Status    Duration
-------------  --------  ------------
LPC1769        SUCCESS   00:01:26.065 
Finely got the stuff to compile now more testing is needed before any bottles are open to mark success but its one less nail and 1000 to go. Bugfix finely  worked for me so maybe there is some luck left. Sorry but i have lost all fate in Chinese products every thing seems half done......

Doyle4, 2020-09-12:
Will have a go at this.
What was your steps to achive this?

Doyle4, 2020-09-12:
Still failing on me, changed over everything from above guide, I get instant failed.

Doyle4, 2020-09-12:
This is ridiculous... If I had known I was buying a board that doesn't accept building a firmware, Id given this a miss

Doyle4, 2020-09-12:
> Okay =) No problem.
> Download this archive: https://github.com/MarlinFirmware/Marlin/archive/2.0.6.1.zip - It's stable Marlin version.
> Unpack archive.
> Download and replace this files: - It's files from Makerbase PR in to Marlin.
> https://raw.githubusercontent.com/mks-viva/Marlin/4bdaec14dbb68006ef9067307143d45ea0080996/Marlin/src/pins/pins.h
> https://raw.githubusercontent.com/mks-viva/Marlin/4bdaec14dbb68006ef9067307143d45ea0080996/Marlin/src/core/boards.h
> https://raw.githubusercontent.com/mks-viva/Marlin/4bdaec14dbb68006ef9067307143d45ea0080996/Marlin/src/pins/lpc1769/pins_MKS_SGEN_L_V2.h
> Path for replace you can see in URLs:
> 
> * Marlin/src/pins/pins.h
> * Marlin/src/core/boards.h
> * Marlin/src/pins/lpc1769/pins_MKS_SGEN_L_V2.h

When you say replace the files, are we replacing the 2.0.6.1 files with PR ones? or other way round, as I have done both ways and still fails, Environment doesn't change to LPC1769 either..

helgjohn, 2020-09-12:
> This is ridiculous... If I had known I was buying a board that doesn't accept building a firmware, Id given this a miss

Got it Working and seem to work haven`t tested for longer time but it moves and heats you know.... I used the bugfix download from official site to marlin and had already re installed VS Studio code and all ext, then i som how got it to work But i think its more luck than knowhow here on my part. but i take it .   
 The thing is that they can make it seamless, but its like but cant be bother some times,  and release products that are half Way finish or a prototype at best, and that is the theme song in more than one product from the fare east. 
So the real kikker is its cheap so it cost less to replace than fix or send back, and you are also in the stigma that you get what you pay for, Like it getting to be a better product if you pay more so the middleman can drive a new Car... just saying ok. so now a little less steam.

Doyle4, 2020-09-12:
@helgjohn Have you got FIL_RUNOUT_SENSOR allocated? I have tried giving it a pin and still says its needs a pin, if not it fails, disabling it compiles fine.

Vaan530, 2020-09-12:
Got Fail compile when EEPROM enable, it's success when I common #define EEPROM_SETTINGS.

helgjohn, 2020-09-12:
> @helgjohn Have you got FIL_RUNOUT_SENSOR allocated? I have tried giving it a pin and still says its needs a pin, if not it fails, disabling it compiles fine.

NO dont use that dont see the point sorry

helgjohn, 2020-09-12:
have now printed my first print with the mks card and seem to be ok no surprises or major artifacts. print is un tuned just whit stander settings. 
[image]

helgjohn, 2020-09-12:
ok guys now closing this issue since it seem to be solved thank you for all respons and comments,

closed 2020-09-12 by helgjohn

Doyle4, 2020-09-13:
Cool, i'll leave filament run out uncommented, eeprom defined is fine for me, but will test soon. Thanks for helping.