makerbase-mks/MKS-SGEN_L-V2 #10: SD Mount

Doyle4, 2020-09-16:
Hey peeps, i have changed serial port 1 to 0 and -1 as well as changing serial port 2, I cant seem to get the SD card to mount when connected to a computer.

Does the board need a psu to do this or can it work the same as SKR boards where a jumper can be placed when not connected to a psu?

makerbase-mks, 2020-09-17:
Hi! 
1. You need set SERIAL_PORT -1 and set SERIAL_PORT_2 0
2. If you use marlin 2.0.x on SGEN_L V2 motherboard and first, you may be need update COM and file in: Marlin->src->HAL->LPC1768->win_usb_driver

Doyle4, 2020-09-17:
Tried these, SD card still not mounting on my mac.

makerbase-mks, 2020-09-18:
Can you try to use smoothieware, because i test it is OK, so i hope you test it.

Doyle4, 2020-09-18:
Never heard of this, will give it a try and will let you know.

Doyle4, 2020-09-18:
SD Card mounted using Smoothieware, so at least my board is fine.
Never used Smootheware so unsure how to go about it, fingers crossed a working stable Marlin will be released soon, had this board over a couple months now and not able to use it, shame the board was pushed so early.

makerbase-mks, 2020-09-18:
You need latest version marlin? I download latest version marlin and test it OK. I think it has nothing to do with the port.

Doyle4, 2020-09-18:
Thank you, will give that firmware a try.
I have tried 2.0.6 which the board is incompatible, i have also tried 2.0.6.1 which has the same issue and have tried bugfix 2.0.X which worked, but SD didnt mount.
Will try your file thanks.

Doyle4, 2020-09-18:
The firmware has worked thank you :) SD now mounts!
Could I ask, is it possible to send me the source to the firmware you provided so I can edit the config/config_adv please?
thank you so much!

makerbase-mks, 2020-09-19:
@Doyle4  I has upload it, you can download it.
[Marlin-bugfix-2.0.x-SGEN_L-V2.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/5248913/Marlin-bugfix-2.0.x-SGEN_L-V2.zip)

Doyle4, 2020-09-19:
@makerbase-mks Thank you so much! Very much appreciated! Late here, tomorrow I will compile and if all good I'll close this :) Thank you again.

Doyle4, 2020-09-19:
All seems good so far, although im having issues when matching to my MKS GEN L 8bit board configs that I cant seem to solve.

#error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
 1392 |   #define Z_SAFE_HOMING_X_POINT X_CENTER ( / 2)    // X point for Z homing
      |                                            ^
Marlin/src/HAL/LPC1768/tft/../../../inc/../core/macros.h:215:45: note: in definition of macro 'WITHIN'
  215 | #define WITHIN(N,L,H)       ((N) >= (L) && (N) <= (H))
      |                                             ^
Marlin/src/HAL/LPC1768/tft/../../../inc/SanityCheck.h:1502:24: note: in expansion of macro 'Z_SAFE_HOMING_X_POINT'
 1502 |   static_assert(WITHIN(Z_SAFE_HOMING_X_POINT, X_MIN_POS, X_MAX_POS), "Z_SAFE_HOMING_X_POINT can't be reached by the nozzle.");

#error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."

I would of thought matching configs would work?

makerbase-mks, 2020-09-23:
Which LCD?

Doyle4, 2020-09-23:
mks tft28

makerbase-mks, 2020-09-26:
If use MKS TFT28, you needn't define ADVANCED_PAUSE_FEATURE , because you can use this function on MKS TFT28 directly

closed 2020-10-08 by Doyle4