makerbase-mks/MKS-SGEN_L-V2 #13: problems compiling in platformio

patxinet, 2020-10-03:
hi, i have problems compiling in platformio the Marlin-bugfix-2.0.x-MKS TS35 + SGEN_L V2,
I want to include the bltouch, with what input, I have to remove babysteping, because it needs a graphic tft, and it gives an error, everything seems to be going well, until this error appears:
c: /users/xavi/.platformio/packages/toolchain-gccarmnoneeabi/bin /../ lib / gcc / arm-none-eabi / 9.3.1 /../../../../ arm-none -eabi / bin / ld.exe: region RAM overflowed with stack
collect2.exe: error: ld returned 1 exit status
*** [.pio \ build \ LPC1769 \ firmware.elf] Error 1
Someone knows what is due?
thanks

patxinet, 2020-10-04:
I think it is solved, the error appears when #define EEPROM SETTINGS is enabled and the screen is defined as #define TFT_480x320_SPI. That combination gives error, you have to choose to have eeprom settings or marlin UI.
#define SPI_GRAPHICAL_TFT and #define EEPROM SETTINGS does not give a problem
This board is not what it seems, it has many limitations.
It's like having a BMW car, and not being able to put electric windows, or air conditioning

makerbase-mks, 2020-10-06:
Thank you for your feedback, we will follow up and send feedback to marlin

doug-scott, 2020-10-26:
> I think it is solved, the error appears when #define EEPROM SETTINGS is enabled and the screen is defined as #define TFT_480x320_SPI. That combination gives error, you have to choose to have eeprom [...]

You said in your first post that you have a MKS TFT35 and the SGEN_L V2.0.  If that is the only hardware involved, I don't think you need to define the MKS TFT35 touchscreen in the firmware.  Try not declaring it all and rebuild the firmware.  You setup the touchscreen by editing the mks_config.txt file and saving it to the root of the SD card you have in the TFT35.  When it powers up it will write the changes to the firmware on the TFT35.

patxinet, 2020-10-26:
"hi, i have problems compiling in platformio the Marlin-bugfix-2.0.x-**MKS TS35** + SGEN_L V2"
MKS TS35 not MKS TFT35
regards

mks-viva, 2020-10-28:
> "hi, i have problems compiling in platformio the Marlin-bugfix-2.0.x-**MKS TS35** + SGEN_L V2"
> MKS TS35 not MKS TFT35
> regards

Yes, you need use MKS TS35 connect to EXP1 EXP2: https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/MKS_TS35

zigou83, 2020-11-26:
Hi, is there a solution to this problem ?
I just received my board and discovered that, it seem unreal that you can't use TS35 in color mode with  EEPROM_SETTINGS together !
Makerbase please do not make me regret this TS35 after the deception of TFT28/32.
Also do you plan to make MKS UI available on SGEN L V2 ?

mks-viva, 2020-11-27:
@zigou83 
Hi! 
1. About EEPROM. 
I has test it. Use MKS SGEN_L_V2+MKS TS35 run the latest version marlin bugfix 2.0.x and EEPROM function is OK. I has upload test firmware bin file, you can download it and test it.
[SGEN_L_V2 Color UI and EEPROM.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/5605520/SGEN_L_V2.Color.UI.and.EEPROM.zip)

2. About MKS UI. 
There is currently no plan to support MKS UI, the main reason is
A. SGEN_L V2 motherboard does not reserve SPI flash and IO is limited
B. MKS TS35 did not consider the use of motherboards other than MKS Robin Nano V2 at the beginning

Thank you for your feedback, we will collect more user needs and feedback.

twinjet75, 2020-12-06:
Hi, I have the same problem. Using MKS TF35 V2.0 in color mode with #define TFT_480x320_SPI and #define EEPROM SETTINGS  generate V Code error during compiling:

Indexing .pio\build\LPC1769\libFrameworkArduino.a
Linking .pio\build\LPC1769\firmware.elf
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1

Please mks-viva can you send your configuration.h and configuration_adv.h? Also I tried to install mks-viva firmware and request monitor calibration at every boot and after storing eeprom sometimes touch does not work anymore.

> @zigou83
> Hi!
>
> 1. About EEPROM.
>    I has test it. Use MKS SGEN_L_V2+MKS TS35 run the latest version marlin bugfix 2.0.x and EEPROM function is OK. I has upload test firmware bin file, you can [...]

mks-viva, 2020-12-07:
Unfortunately, I did not save it. But you can provide your firmware to me and i will check it.

dlaval1, 2021-01-20:
Bonjour
J'ai une MKS Gen L V 2.0 avec des TMC 2209 que je souhaiterais mettre sur une ender 3 pro.
J'ai suivi les consignes mais j'ai un problème de construction avec platformio 

> Executing task: C:\Users\HP\.platformio\penv\Scripts\pio.exe run <

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)        
--------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
ModuleNotFoundError: No module named 'platformio.managers.package':
  File "C:\users\hp\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript(item, exports="env")
  File "C:\Users\HP\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\HP\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\PERSO\3D\CARTE MERE\FIRMWARE\MKS-SGEN_L-V2-master\Firmware\Marlin-2.0.6\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 12:
    from platformio.managers.package import PackageManager
=========================================================== [FAILED] Took 0.78 seconds ===========================================================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:00.779
===================================================== 1 failed, 0 succeeded in 00:00:00.779 ===================================================== 
The terminal process "C:\Users\HP\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Merci d'avance
Dominique

mks-viva, 2021-01-21:
@dlaval1 Hi! I think you need upload your firmware to here and i will download test it.

dlaval1, 2021-01-21:
> @ dlaval1 Salut! Je pense que vous devez télécharger votre firmware ici et je vais le télécharger pour le tester.
Bonjour
Voici  le lien pour le firmware
https://drive.google.com/drive/folders/1XGodXCRxUzrTq7shkb2a0rNfJF3eO1hK?usp=sharing

Merci pour votre aide.
J'avais une question, puis-je ajouter un MKS TFT35 avec cette carte ?
Cordialement
Dominique

dlaval1, 2021-01-22:
@mks-viva Bonjour
voici le lien 
https://drive.google.com/drive/folders/1XGodXCRxUzrTq7shkb2a0rNfJF3eO1hK?usp=sharing
merci
dlaval1

mks-viva, 2021-01-25:
@dlaval1
I tried to compile it, but it failed. I think it was caused by the platfromio update. You can try the new version firmware.

dlaval1, 2021-01-25:
@ mks-viva 

> @ dlaval1
> J'ai essayé de le compiler, mais cela a échoué. Je pense que cela a été causé par la mise à jour de platfromio. Vous pouvez essayer la nouvelle version du micrologiciel.

Je suis sur la dernière version de Platformio et j'ai pris la dernier version que vous avez mis en ligne via aliexpress. Pourriez-vous donc joindre la dernière version que vous dites. merci beaucoup.

mks-viva, 2021-01-26:
@dlaval1 
MKS SGEN_L V2 path firmware is not the latest, you need download the latest marlin firmware by Marlin https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

closed 2021-03-06 by mks-viva