makerbase-mks/MKS-SGEN_L-V2 #15: MKS SGEN L V2 + Fysetc Mini12864

LeandroMarceddu, 2020-10-05:
Hi. I've been using this combo and I'm trying to figure out the pins and how to get it to work. 
I'm not that good in electronics, so if you would be able to help me that would be good.

makerbase-mks, 2020-10-06:
Can you provide Fysetc Mini12864 pin map?

LeandroMarceddu, 2020-10-06:
> Can you provide Fysetc Mini12864 pin map?

Yes I can. https://wiki.fysetc.com/Mini12864_Panel/

LeandroMarceddu, 2020-10-07:
Fixed it myself.

Just plug the connectors in the right way. These are the pins:

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P1.31, EXP1_3=P0.18, EXP1_5=P0.15, EXP1_7=P1.0,  EXP1_9=<GND>,
    EXP1_2=P1.30, EXP1_4=P0.16, EXP1_6=P0.17, EXP1_8=P1.22, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.8, EXP2_3=P3.25, EXP2_5=P3.26, EXP2_7=P0.27, EXP2_9=<GND>,
    EXP2_2=P0.7, EXP2_4=P0.28, EXP2_6=P0.9,  EXP2_8=<RST>, EXP2_10=<NC>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp1"

closed 2020-10-07 by LeandroMarceddu