makerbase-mks/MKS-SGEN_L-V2 #16: Extruder fan not working

Doyle4, 2020-10-08:
I did a 1:1 copy of my configs for the MKS GEN L board and changed steppers and directions etc, but for the life of me can not get the Extruder fan to work, does anyone know where to change the pin or define a new pin please? it use to kick in around 50c automatically. 

Thanks!

Doyle4, 2020-10-08:
#define E0_AUTO_FAN_PIN 7 is what my MKS GEN L was set too, and what the SGEN L V2 is set too, but its not working.

In the pins.h file there is no mention of pin 7 for the old board or new, pretty confusing

Doyle4, 2020-10-08:
got it, P2_06 was needed, not 07

closed 2020-10-08 by Doyle4

ILGROSSOFA, 2021-02-06:
> got it, P2_06 was needed, not 07

Hi Doyle4, 
I have the same problem on my MKS SGEN L v2: where did you change value to P2_06? Thank you in advance!

Doyle4, 2021-02-06:
@ILGROSSOFA 
Configuration_adv.h > #define E0_AUTO_FAN_PIN P2_06

ILGROSSOFA, 2021-02-06:
> @ILGROSSOFA
> Configuration_adv.h > #define E0_AUTO_FAN_PIN P2_06

Now there is written:
#define E0_AUTO_FAN_PIN 7
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define E5_AUTO_FAN_PIN -1
#define E6_AUTO_FAN_PIN -1
#define E7_AUTO_FAN_PIN -1
#define CHAMBER_AUTO_FAN_PIN -1

#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed
#define CHAMBER_AUTO_FAN_TEMPERATURE 30
#define CHAMBER_AUTO_FAN_SPEED 255

Is it correct to change from 7 to P2_06?

ILGROSSOFA, 2021-02-06:
It looks like I have to add following command in pins_MKS_SGEN_L_V2.h file:
#define HEATER_1_PIN                       P2_06
Isn't it?

ILGROSSOFA, 2021-02-07:
SOLVED!
In configuration.adv.h file:
#define E0_AUTO_FAN_PIN P2_06

dplumly, 2022-01-22:
> 

Hi ILGROSSOFA,

I am having the same issue, but just defining: #define E0_AUTO_FAN_PIN P2_06 in configuration_adv.h throws and error - P2_06 not declared in scope. Is there something you needed to declare first? Thank you.

ILGROSSOFA, 2022-01-31:
Are you using Arduino or Visual Studio Code.
I have used Visual Studio Code and it worked for me.

dplumly, 2022-01-31:
> Are you using Arduino or Visual Studio Code? I have used Visual Studio Code and it worked for me.

I have used both. The way I did get it to work for anyone else that has this issue is I didn't use the terminal that said fan. In configuration_adv.h, I set my fan to E0_AUTO_FAN_PIN 7. I then took 2 jumpers and plugged the fan into E1 interface.

truonghuutoanonenote, 2022-07-21:
simply just plug the fan to 12v output on top left conner and everything is done

ezpzlmnsqz1337, 2025-11-20:
> I have used both. The way I did get it to work for anyone else that has this issue is I didn't use the terminal that said fan. In configuration_adv.h, I set my fan to E0_AUTO_FAN_PIN 7. I then took [...]

Indeed, define E0_AUTO_FAN_PIN 7 and use the E1 terminal, not the always on connector next to the stepper drivers.