makerbase-mks/MKS-SGEN_L-V2 #18: Options for using FES and 3Dtouch that want to use the same pin

doug-scott, 2020-10-09:
I received my MKS SGEN_L v2.0 the other day.  I am trying to configure the two components to work together.  Both pieces say to use pin P1_24, and to connect both devices to the same socket on the mainboard.  The videos do not directly mention the v2.0 motherboard, nor do they show having both devices working on the same board at the same time for any of their boards.  

I am not large on knowledge, but, in my simple way of looking at it, what is to stop me from using the Z- Endstop connector for the FES, using pin P1_24, and setup the way it shows in the video(except the part that leads you to the wrong file, but use the 1769 version of the pins file).  Then go follow the 3Dtouch video, and again recognize the different names for headers, and connect the two wire connector to the Z+ Endstop connector and use pin P1_25 where it says to set it to P1_24.  Would that work, or am I missing a lot of other changes that would need to be done?

The instructions on WIKI for 3Dtouch are somewhat confusing at the end.  It says:

3D Touch replace z-axis endstop
According to the "3D Touch replace z-axis endstop" requirements Connect motherboard

Firmware setting(after normal firmware setting)

Configuration.h
Enable: #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
Enable: #define Z_SAFE_HOMING
Disable: //#define Z_MIN_PROBE_PIN P1_24

To me this says basically after you get it working, rebuild the firmware again after making those 3 changes, then install and reboot.  Is that what it means?  Can't see how that would even be required.

The online videos are great to have, but, trying to follow the order it is presented in does not work.  How is he managing to get it to work BEFORE he made changes to configuration files?

makerbase-mks, 2020-10-10:
marlin+3D touch : https://github.com/makerbase-mks/MKS-SGEN_L-V2/tree/master/Firmware/3D%20TOUCH

doug-scott, 2020-10-11:
Thanks for the updated firmware source.  I downloaded the marlin+3dtouch archive this morning and compared the configuration files to the ones I have for Marlin 2.0.7.  There are a couple of things I saw that may or may not be issues, 
1.  In the configuration.h you have "#define TEMP_SENSOR_BED 1" instead of "#define TEMP_SENSOR_BED 0", is this correct for this board?  I can not see any reference to a second BED temp sensor on the physical board.
2.  The youtube video for setting up the 3DTOUCH sensor shows a set of settings that use outdated terms and says to enable them, It looks like this in the configuration_adv.h in the archive:

#if PROBE_SELECTED && !IS_KINEMATIC
  //#define PROBING_MARGIN_LEFT PROBING_MARGIN
  //#define PROBING_MARGIN_RIGHT PROBING_MARGIN
  //#define PROBING_MARGIN_FRONT PROBING_MARGIN
  //#define PROBING_MARGIN_BACK PROBING_MARGIN
#endif

The video says to enable these settings 
#if PROBE_SELECTED && !IS_KINEMATIC
  #define PROBING_MARGIN_LEFT PROBING_MARGIN
  #define PROBING_MARGIN_RIGHT PROBING_MARGIN
  #define PROBING_MARGIN_FRONT PROBING_MARGIN
  #define PROBING_MARGIN_BACK PROBING_MARGIN
#endif

I can build it without error using either setting.  I am currently using the 2.0.7 source.  2.0.6 had a fatal error that was fixed in v2.0.6.1.  I don't recall what the issue was, but I know it caused a fatal issue on my setup.  Is the version in the archive running 2.0.6.1?

My system is still being put together, I decided to redo the setup to make it easier to use and service, quickly finding out that any time I will save after doing this will never recover the time it is taking to make it easier.

makerbase-mks, 2020-10-12:
We will fix it

commit a5258f49 (2020-10-14): 3D TOUCH firmware fix #18
closed 2020-10-14 by mks-viva