makerbase-mks/MKS-SGEN_L-V2 #19: On-board SD card not recognized: unable to install/update firmware

NeoMod, 2020-10-10:
I've bought a **brand new SGEN-L v2** last week and today I've finally found some time to mount it on my printer, except I've found out that I **can't install/update the firmware** because **the on-board microSD isn't recognized**.

I've tried both under Windows and Linux: **the board is recognized but not the microSD**. Under **Windows** It sometimes appears as a **dummy drive**: the enumerator assigns a letter to the drive but I'm **unable to access** it. Almost as If I had connected a card reader with no card inside.

I tried resetting the board using the _on-board reset button_ but apparently, this switch only causes a soft reset.

One strange thing I noticed: **this board had already a firmware installed** (Marlin), with some customizations, _while with the previous_ (two SGEN-L boards) _I didn't find any pre-installed firmware._ 

_(and yes, I'm sure the Marlin installed on the board is not my compiled version: my customizations are missing and there are some which I've never used.)_

**I also tried with the M997 command** with no luck: the board enters a sort of halt state _(power LED "red" always-on, other LEDs off)_ and I **still can't access the microSD**. 
I've also tried renaming the microSD to "REARM" and creating a dummy "firmware.cur" file on it, hoping to trigger the recognition but to no avail.

I hope there is any possible solution to this issue aside buying a new board.

NeoMod, 2020-10-10:
Update: I suspected the previous configuration (probably the one the board was tested with) had a different LCD connected with a different setting for SD usage.

I've been able to release the on-board microSD from Marlin access with the "M22" command and - at least - I can confirm the microSD is working. I can access it via USB just as expected.

The issue is still the same, however: I can't update the firmware to correct this situation. In order to upgrade the firmware I need to power-cycle the board, but as soon as I do that the microSD is locked again and the board won't acknowledge the firmware.bin file on it. Feels like an endless circle.

NeoMod, 2020-10-11:
Ok, so after pulling an all-nighter on this I finally found a way to solve the issue.

There are **two possible roads to follow:**

1. **Re-Flash the LPC1769 Bootloader** (FTDI Cable or Arduino) and start fresh with the firmware.

2. **Follow a specific sequence of actions** _(don't know if specific only to this board or configuration)_ to "force" the FW Update.

Before continuing, let me point out **a couple of things I've learned**.

- **There are many similar issues scattered along with internet regarding LPC176x and Bootloader issues;** it seems not so uncommon to face a somehow corrupted bootloader leading to such problems.

- **Re-flashing the bootloader is way to much trouble** (especially if you are not familiar with this kind of things) **but it's the best course of action.** After solving my issue with the 2nd method I went ahead and re-flashed the bootloader: all stability issues disappeared.

- Sounds obvious, but...**before diving into this rabbit hole check first for the basic issues**: SD Card, USB Cable, Power Source, Connections, etc.

I'm not going to describe how to flash the bootloader, there are plenty of guides available and the LPC176x community has some nice tutorials. 

As for the second method, **here is what tricked the successful firmware update**:

- Connect the board to the printer in "production" mode, meaning all components connected and wired.

- Be sure to have the **LCD connected** _(this is important, as I suspect the whole heart of the problem comes from testing the board with a REPRAP LCD instead of a 12864 LCD)_

- **Disconnect the TFT TouchScreen** from the board. No matter the model, keep it disconnected from AUX-1.

- **Remove the microSD** from the board and power it up via USB Cable.

- Board will boot with the previous configuration (you can verify with M502/M501) 

- Issue an **M22** command, disconnect from the board (if you are using pronterface use the disconnect button) and power off.

- **Prepare the microSD** card (FAT32 filesystem, less than 32GB) with two files: **firmware.bin** _(the actual firmware you are going to install)_ and **firmware.cur** _(an empty file)_**. Rename the card as "REARM"** (not sure if needed, but I did).

- Insert the card into the board, and **power up via USB**. Almost certainly the board will start, once again, with the old firmware.

- Issue an **M22** command, wait until the microSD is accessible from your computer, and then issue an **M997**.

- **The board will reboot and starts a "forced" update.** The first time you may end up with a RED LED always-on and no communication with the board: power off and repeat the previous two steps. Don't be too quick between reboots otherwise the board will start in "smoothieware" status and you'll have to start all over.

Hopefully, in the end, **the board will start updating the firmware upon reboot** _(you can notice the BLUE LED pattern changing)_ and it will reboot; once the LCD shows the Marlin Bootscreen the procedure is successful. Wait about 1 minute for the start-up to finish and confirm EEPROM INIT via LCD. 

Good luck with your board, I hope this could help you not wasting an entire night on a trivial issue such as this one.

closed 2020-10-11 by NeoMod

michalfrnoch, 2021-05-23:
please, link to bootloader.bin

Sviatoy, 2024-12-13:
This works for me, thank you!
`- Be sure to have the LCD connected (this is important, as I suspect the whole heart of the problem comes from testing the board with a REPRAP LCD instead of a 12864 LCD)
- Issue an M22 command, wait until the microSD is accessible from your computer, and then issue an M997.
`