makerbase-mks/MKS-SGEN_L-V2 #20: Unable to use both z-axis ports

doug-scott, 2020-10-13:
When I try to use the 2 z-axis ports on the SGEN_L V2.0 board the stepper motors end up fighting for control.  Its like one goes left, the other goes right.  If I shut the printer off and unplug the power to the printer, then turn one of the z-axis stepper by hand, the other stepper motors turns by itself the opposite way.  I would assume this is due to some induction thing.  

I have gone through all the stepper motor wiring(the steppers I have came with 4 wires coming out of the motor in a bundle, and at the end of the wire it had a simple flat 4 wire terminal) and changed one to match the rest(the y-axis was set to reverse, now it is same).  Tried swapping motors position, the issue moved with the change(ie I took x-axis connector and put it in z1 port, x-axis stepper now misbehaves).  The last thing I did was go back to the standard of using the 2 into 2 stepper motor wire adaptor and used only the one z-axis port on main board.  Now z-axis no longer has the induction issue, but, still does not work correctly.  

I have attached my config files.  I am running the v2.0.7 Marlin-bugfix-2.0.x.

[Configuration_files.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/5374046/Configuration_files.zip)

mks-viva, 2020-10-14:
Can you provide your motherboard wiring ? If have TMC CONNECTION ERROR, you need check your DC12-24V power is power on?

doug-scott, 2020-10-14:
I am not sure what you mean by "provide your motherboard wiring". 

I have the 5 MKS TMC2209 drivers plugged in, and defined in the configuration.h.  These drivers are only plugged in, no jumpers moved or installed.
All three axis move.

The issue with using the 2 z-axis ports is still on-going.  I have stopped trying to use the 2 ports and am using the original 2 into 1 z-axis adaptor and plugging it into the Z1 port.  I am assuming that the reason for the 2 ports was to remove the need to use that adaptor.  Am I wrong in assuming that?  Is there some configuration change needed in order to use the two z-axis ports?

mks-viva, 2020-10-14:
If you use Z1 and Z2 interface on the board, you needn't set it. If you use E1 as Z2, you need set Dual Z axis, like this: 
[image]

doug-scott, 2020-10-14:
Would using E1 as Z2 give you separate TMC2209 drivers on each of the two ports, as opposed to sharing one TMC2209 driver if you just used Z1 and Z2?

mks-viva, 2020-10-15:
You need #define Z2_DRIVER_TYPE TMC2209

doug-scott, 2020-10-15:
Just going to update this topic, In a separate discussion I was told that if you want to use the two z-axis ports, you cannot have more than the one extruder.  I don't recall how you enable the second z-axis port to use the E1 driver though.

mks-viva, 2020-10-16:
If has use E1 axis as Z2, so, you only one extruder mode, if you want to use two extruder, you need Z2 interface on the board
[image]

doug-scott, 2020-10-18:
> If has use E1 axis as Z2, so, you only one extruder mode, if you want to use two extruder, you need Z2 interface on the board
> 
Is adding "#define Z2_DRIVER_TYPE TMC2209" the only change required?  I get a "TMC Connection Error" when I have driver_types set to TMC2209.  If I comment those lines out, the error message goes away.  I am using the MKS TMC2209 v2.0 drivers.  I was told by support to comment out the lines to correct the error.  Have not back about what functionality do I lose by not configuring the drivers in the driver_type declaration.  I am thinking I just purchased something that does not work on the board they were packaged with.

mks-viva, 2020-10-19:
If you get a "TMC Connection Error":
1. jumpers connect to UART mode on the board
2. You need use Dc12-24V power input

doug-scott, 2020-10-19:
Which jumpers and to where?  Or is there some online documentation that describes this procedure, maybe a manual for the motherboard?

I do not understand what you mean by "You need use Dc12-24V power input".  The only way I can take that is to mean I need to supply the motherboard with 12/24v DC.  Although, if I didn't have that already done, nothing would work beyond what the mainboard can get from a PC connected via the USB port.  So, I truly have no idea what you mean by "You need use Dc12-24V power input".

mks-viva, 2020-10-20:
You can refer to Wiki: https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/Drivers_MKS_TMC2209
Setting UART mode:
[image]

mks-viva, 2020-10-22:
@doug-scott 
Is this problem solved?

doug-scott, 2020-10-22:
Thought I had posted this already, guess not.

After setting the jumper, as it is shown in the pic posted here, for all 5 drivers, the TMC CONNECTION ERROR no longer appears on bootup of the 12864 device.  Thank you.
I will close this issue.

closed 2020-10-22 by doug-scott