makerbase-mks/MKS-SGEN_L-V2 #21: Is "LED1-4" PIN SET tied to the BOARD STATUS LEDs ?

NeoMod, 2020-10-16:
Looking at the pinout, one of the advantages of this board was - supposedly - having four dedicated PINs to devote an RGB LED. 

Those PINs (P1_18, P1_19, P1_20, P1_21) are labelled as "LED1-4" in the Pinout Scheme, but in the Wiring Scheme they are marked as "Reserved".

Upon testing, I discovered that those PINs are indeed tied to the motherboard status LEDs (bottom left corner, near the USB). These LEDs are not present on the Pinout Scheme, but the silkscreen identifies them exactly like the above. 

Empirically defining those pins for an RGB LED shows that they are indeed connected. Assigning a blue led (actually, any led. colour doesn't matter) to the P1_18 pin and thus defining it as STAT_LED in Marlin, shows that this led will keep blinking in sync with the motherboard status led. 

Am I getting something wrong? Or those pins are indeed meant as an extension port for the motherboard status LEDs?

mks-viva, 2020-10-16:
Hi! LED1,P1_18 use debug led. But you can define LED2-4 pin to other function, because LED2-4 have not function in marlin
[image]
[image]

NeoMod, 2020-10-16:
Thanks a lot! I totally missed that scheme.
Great, I'll move to P1_19/_20/_21 as suggested for the RGB LED, and now since I know about P1_18 I'll probably use it to have a status indicator outside the printer frame. 
Thanks again for your help.  👍

closed 2020-10-16 by NeoMod

mks-viva, 2020-10-16:
Ok! We will submit this functional requirement to marlin and Using the RGB function will become easy on SGEN_L V2.