makerbase-mks/MKS-SGEN_L-V2 #22: Auto level 3D touch and MKS S gen L v.2

helgjohn, 2020-10-22:
Have an issue with 3d touch sensor The same issue is also present if I use any SKR board or MKS series board. As far as I have found it has the same issue on skr 1.4 all ver .- 1.3 all ver.  And MKS SgenL (ver 1 and 2) running MARLIN firmware . 
Have current marlin ver 2.0.x  not sure if same issue are present on older marlin ver. 
Here is what happens : 
-The probe does its thing and collect all data about the bed and so on, but does not use the collected data in order to adjust the nozzle when printing starts Have testes different values of fade high no change in behavior  . (the auto level function is not working)   
-So the ting i have now is a advanced end stop. 
Here is the tings I have done:
In order to get it up and running new install was done i now run at marlin ver 2.0.7.. 
-Have also tested with new board / sensor and Wire ring and connectors it make NO DIFFER. 
-Have also tested all settings one by one also in diff combo and the only thing that works in the end is new install of MARLIN whit same setup as before. But after a while the auto level function stop working at any level  and a new install of Marling seem to fix it for a while. 
-If I activate Dual X on separate drivers and auto align it works fine AND some times it doesn’t  but then the homing also misbehave and restart is the way to get it running. 
 so i conclude there is an issue with software and board Sensor combo. 
But I have no luck pinpointing it yet.

The video for installment by MKS is a bit messy and to me, the video seem to be about marketing more products and some surface detail on how to setup/installing the product.  
but i have to the best of my ability followed the steps, there is no video for MKS SGenl v2 (1769 chip) so i guess the instructions are the same the same as Sgenl (1768 )

mks-viva, 2020-10-22:
[image]

mks-viva, 2020-10-22:
You can refer to it.

doug-scott, 2020-10-22:
> Have an issue with 3d touch sensor The same issue is also present if I use any SKR board or MKS series board. As far as I have found it has the same issue on skr 1.4 all ver .- 1.3 all ver. And MKS [...]

How are you determining that it is not following the auto-level data?  I have not tried to check it on the SGEN_L V2.0 board, but did check it on a SGEN board I had.  The SGEN board is an older LPC1769 board.  What did to test it was create a test print by taking a file from the All-in-one box I am using and scale it up to use most of the bed.  This object is a simple 2mm thick plate.  Once it starts printing I would take a piece of masking tape and stick it to the z-axis stepper motor chassis and up to the isolator that connects the rod to the stepper motor.  If the stepper motor moves as it should to account for a bed that is not 100% level, the tape would pull off the stepper.  I didn't really care if it was accurate, I was simply looking to see if it at least made the attempt.  I will try it again later today with my SGEN_L V2.0 and post the results.  As an aside note, I also make sure the bed is not level by moving one of the levelling screws a turn or two.

helgjohn, 2020-10-22:
Well the part where auto leveling dont work shod be easy to see the z axis dont move except for layer change and the part where i am sure its reading the different hige and low spots on the bed you get by monitoring software like pronteface or on the screen on you printer if you have enabled the right stuff in marlin. its called mesh something....

doug-scott, 2020-10-22:
> Well the part where auto leveling dont work shod be easy to see the z axis dont move except for layer change and the part where i am sure its reading the different hige and low spots on the bed you [...]

Ok, I lowered one corner of my bed, ran the G29, then M500 then printed that GCODE file I made.  Didn't need the tape, I could see it moving both z-axis steppers.  I put my fingers on the flexible shaft couplers for the entire first layer (easier to feel it than to stare at it) then ignored it for 2nd and 3rd layer, went back to touching the couplers for about half the 4th layer.  It was still adjusting at the 4th layer.  

I don't see the connection between the 3dtouch sensor and how it auto adjusts while printing.  The sensor is use to create the "database", and the operating system uses that table to adjust the bed while printing.  The 3dtouch device just goes along for the ride.  It is used at the very beginning of the print job to establish where it physically is, but it then is not used.  When you run G29 you can run M503 and see the locations it checked.  I run the M500 after G29 just to make sure the settings are saved before the G28 gets run.  I have a feeling that the G28 does something to what the G29 measured, so I save it just to be sure.  

I am not a fan of the videos either, they try too hard to use as little text as possible on as many boards as possible.  They try, but, they are simplifying the wrong areas.