makerbase-mks/MKS-SGEN_L-V2 #23: MKS LCD12684B V1.0 not working

Bencr2, 2020-10-24:
Hello,

I have a MKS SGEN_L V2.0 together with a MKS LCD12684B V1.0.

My Problem is, that the display is not displaying anything currently and therefore im not even sure if the flash was succsesfully buy the SD Card.

After restarting the board with the Firmware.bin on the SD Card the file on the SD Card will be remanemd in a .cuorser File. 

The LCD backlight is on
The beeper is working

I used the latese Marlin 2.x bufix and changed:
PlatformIO.ini to default_envs = LPC1769
cofiguration.h to #define MKS_LCD12864
configuration.h #define MOTHERBOARD BOARD_MKS_SGEN_L_V2

Was my flash succesfull?
What do I have to do to get the display working.

Thank you very much
BR
Jochen

mks-viva, 2020-10-24:
The lastest version firmware should is fix it and it is working on MKS SGEN_L V2.
Can you provide your config to me?

doug-scott, 2020-10-24:
If the firmware flashed successfully the firmware.bin file will now be called firmware.cur on the SD card.  The "cur" part just means "current".  

If you have the MKS LCD12864B it should be defined as "MKS_MINI_12864" according to this line in the configuration.h file.

MKS LCD12864A/B with graphic controller and SD support. Follows MKS_MINI_12864 pinout.

mks-viva, 2020-10-27:
Yes, it is compatible with MKS MINI12864.

mks-viva, 2020-10-30:
I will close it

closed 2020-10-30 by mks-viva