makerbase-mks/MKS-SGEN_L-V2 #24: 12864 display not showing Babystep Z when Babystep X and Y are activated

doug-scott, 2020-10-24:
I have enabled Babysteps for x, y, and z axis.  When I use the 12864 to adjust Babystep on the Z axis it is not listed.  It shows "Probe Z Offset".  When X and Y Babysteps are not enabled, it shows "Babystep Z".
My 12864 is defined as "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" in configuration.h.

[image]
[Configuration.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/5433706/Configuration.zip)

mks-viva, 2020-10-27:
I think babystep Z is necessary, babystep X or Y is not very necessary.

doug-scott, 2020-10-27:
I would imagine it was discussed at some time between developers, and they decided that there was a need for it, just not everyone.  I am simply pointing out that there is a bug in there, either in the hardware that may have a max number of fields for babystep labels, and it is one short of what is supported.  Or, a developer never considered the existence of babystep-x and y.

Elzariant, 2020-12-16:
Actually, your Baby-step for Z is activated : "Probe Z offset" is Z-baby-step combined with M851 (means the value is stored in EEPROM for next print). 

`#define BABYSTEP_ZPROBE_OFFSET // Combine M851 Z and Babystepping` 

Note : Personnaly I agree with mks-viva about the usefulness of baby-stepping X or Y during a print.

doug-scott, 2021-01-10:
I think that this issue can be closed.  It was more of an understanding issue than anything else.

closed 2021-01-10 by doug-scott