makerbase-mks/MKS-SGEN_L-V2 #25: Unable to access SD card from PC via USB

doug-scott, 2020-10-26:
Running Marlin 2.0.7.2 on SGEN_L V2.0 with 12864 LCD and MKS TFT35.

Used to be able to compile and upload firmware from my PC to the SD card in motherboard's reader via USB.  After moving to this motherboard I am unable to see the SD card on my PC.  When I was following the WiKi guide is said to enable SDSUPPORT in the firmware.  I did that and reflashed the firmware with that setting.,  That broke being able to see the SD card in the motherboard from my PC via USB.
If I disable SDSUPPORT I can then see the SD card on my PC via USB and can also build and upload the firmware directly to the motherboard.  But, then I can no longer print from the SD card in the motherboard via the 12864 lcd.

mks-viva, 2020-10-27:
I have test it and it is unable to access SD card from PC via USB and i has given feedback to the marlin team.

mks-viva, 2020-10-28:
@doug-scott 
The current solution is that you can access the TF card from the LCD Release Media after the PC via USB.

doug-scott, 2020-10-30:
I am unable to try this as my MKS SGEN_L V2.0 mainboard decided to go up in smoke all on its own the other night.  This board was about a month old at the most, was in a case, sealed from physical access from both below and above the board,  in a printed PLA case, held together with nylon fasteners, mainboard mounted with nylon fasteners, and every step made to ensure no metal was anywhere near underside of mainboard.  MakerBase says this can only happen by pins shorting out.  I have no way of testing for a bad TMC2209 driver failing, causing this, since I have no board I am willing to sacrifice having suffered the same fate, only to be told I must have shorted the board out myself magically, or that the shorting out of the first mainboard must have damaged the driver since their products don't fail due to defects.  I am currently reviewing my options with different manufacturers.  The reason I bought all MKS products, from MakerBase, was to avoid any issues of incompatibility, I never considered the user also having to be compatible with the manufacturer.   

Its a shame that technical support's first and only response insists that the ONLY way it can fail is by user error or fault.  When presented with an alternate theory, they go silent.  People, be wary of their products, MakerBase feels that their products are 100% perfect, regardless of what happens.  Just because they have not seen it happen before only means they have not seen everything.  Really have to wonder how they can claim a brand new designed mainboard that has been in full release for under 4 months cannot possibly have a defect.  If MakerBase wish to discuss this further, feel free to convince me otherwise.

mks-viva, 2020-11-03:
I have already asked the salesperson that you have burnt out two motherboards in a row. I have to doubt whether it is caused by the failure of the external device. We also hope to help you solve the problem.

doug-scott, 2020-11-03:
I went ahead and sacrificed my old MKS SGEN_L v1.0 board by installing it into the box, with the same MKS TMC2209 drivers installed.  Turned power on, and it instantly started smoking at the exact same spot on the mainboard.  I shut the power off right away.  The mainboard is no where near any metal, 

The salesperson still says this cannot happen like that.  He needs to tell the motherboard that.  This was exactly why I purchased all parts from MakerBase.  I wanted to avoid the finger pointing that goes on when you mix components, it is always the other guys problem.  I didn't think the same thing would still happen, except now I am the other guy.  I don't expect to be compensated for the second motherboard failing, it was choice to do that, I thought it may help MakerBase figure out what was happening.  Now I am out the v2.0 motherboard, and the 5 drivers(I do not know which one did it, can't see any burn marks, but one does have a nasty smell to it), plus my old motherboard.  

I bought two of the 3DTOUCH probes at the same time as well.  I was cleaning up the area the other day and connected the cable to the probe and noticed the green wire was popping out of the connector, took it back off, pulled the wire out to pull the lock tab up a bit and while putting it back on something didn't look right.  Found the video on how to install it, and watched it.  The connector on the harness is wired backwards.  You can't put the connector into the probe the other way around, it only fits one way.  I mentioned this to the salesperson and was told that customer should be checking this as part of the installation.  He went on to say that if it was installed backwards it shouldn't damage anything as there is a system on the mainboard that will protect the mainboard from any damage due grounding out the 5v side.  

Makes me wonder what happens if a TMC2209 driver fails and grounds out the 5v side, what protects the mainboard from any damage?  I understand that I "should" see a burn mark on the driver, but, what if it did not short out long enough to burn the driver, but just long enough to burn out that part that did burn out on my mainboards?  The salesperson just ignores that point when brought up.  You would think the $40CAN is coming out of his pay cheque.  

If one person reads any of this, and uses this to help him/her to decide what product to purchase, MakerBase loses that sale, and it does not cost them because they did not back me up, but every one after that comes right out of their pocket.

doug-scott, 2020-12-07:
Makerbase sent me a replacement motherboard that I received the other day.  I installed it in the exact same location, using the same wiring, but this time not installing the TMC2209 drivers.  I had a brand new set of MKS TMC2208 drivers that I installed after setting the board to use UART via jumper under the drivers.  Powered the board up and it powers up without any smoke.  I don't know I can check the TMC2209 drivers to see if there is one or two that have something wrong causing the board to burn that one "thing" in the centre.  I am just going to toss them out,

I am having an issue with 3DTOUCH and z_min being always triggered, I have opened an issue on that specifically already.  

I also got a TFT35 v3.0 display to use instead of using two individual displays, and am having issues with getting it to connect to the printer.  I am dealing with that issue with the vendor of the display.  I am hopeful that using that display will allow me to use it without having to enable SDSUPPORT so that I can connect my laptop when something needs to be updated, adjusted or debugged.  

I think that this issue can be closed with the suggested answer/work-around of mks-viva to use the release option for the sd card in the 12864.

closed 2021-01-22 by mks-viva