makerbase-mks/MKS-SGEN_L-V2 #26: Encoder knob working wierd on LCD12864B with MKS SGEN L V2

Voxel3DPrinting, 2020-10-28:
Hey,
I recently bought an "MKS SGEL L V2" motherboard with the "LCD 12864B"
I configured and uploaded Marlin 2.0 and it works fine, but the encoder knob on the LCD only seems to react when I turn it quickly, if I turn it slowly it just does not register. I already tried adjusting the "#define ENCODER_PULSES_PER_STEP 1 // Increase if you have a high-resolution encoder" value in "configuration.h" but it does not seem to work. Also, the SD card reader does nothing. The "#define SDSUPPORT" is uncommented

Anyone any other ideas?

mks-viva, 2020-10-30:
Hi! Can you provide your marlin firmware to me?

Voxel3DPrinting, 2020-10-30:
Sir,

My marlin firmware is provided here: https://we.tl/t-tTgQqnUE9w I also noticed a problem while homing. The axis just move a tiny bit to the positive side, then stop… what could this be, I provided a video in the link above?

I hope to hear from you soon

Kind regards, 
Sander Hesp

Van: mks-viva
Verzonden: vrijdag 30 oktober 2020 02:04
Aan: makerbase-mks/MKS-SGEN_L-V2
CC: Voxel3DPrinting; Author
Onderwerp: Re: [makerbase-mks/MKS-SGEN_L-V2] Encoder knob working wierd onLCD12864B with MKS SGEN L V2 (#26)

Hi! Can you provide your marlin firmware to me?

mks-viva, 2020-10-30:
Hi! You can #define ENCODER_PULSES_PER_STEP 2, and i test it OK!
[TEST.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/5464607/TEST.zip)

Voxel3DPrinting, 2020-10-30:
IT WORKS!!! Thank you so much!
However, I still have the problem that the SD card is not detected at all. And it does not home. (As shown in the Video of last mail). Do you know what could cause this?

I hope to hear from you!
Sander Hesp

Van: mks-viva
Verzonden: vrijdag 30 oktober 2020 11:52
Aan: makerbase-mks/MKS-SGEN_L-V2
CC: Voxel3DPrinting; Author
Onderwerp: Re: [makerbase-mks/MKS-SGEN_L-V2] Encoder knob working wierd onLCD12864B with MKS SGEN L V2 (#26)

Hi! You can #define ENCODER_PULSES_PER_STEP 2, and i test it OK!
TEST.zip

mks-viva, 2020-10-30:
If you want read LCD's SD card, you need set it in pins_MKS_SGEN_L_V2.h file
[image]
[image]

Voxel3DPrinting, 2020-10-30:
Okay, and the Homing issue?? What could that be?
Thankyou by the way for the amazing help so far!!
Van: mks-viva
Verzonden: vrijdag 30 oktober 2020 12:06
Aan: makerbase-mks/MKS-SGEN_L-V2
CC: Voxel3DPrinting; Author
Onderwerp: Re: [makerbase-mks/MKS-SGEN_L-V2] Encoder knob working wierd onLCD12864B with MKS SGEN L V2 (#26)

If you want read LCD's SD card, you need set it in pins_MKS_SGEN_L_V2.h file

mks-viva, 2020-10-30:
I want to know your endstop is normally open or normally closed and i think you should change endstop logic setting in configuration.h file
[image]

Voxel3DPrinting, 2020-10-30:
I will try soon. Can I mail you later today?

It is really nice to have this feedback!

Van: mks-viva
Verzonden: vrijdag 30 oktober 2020 12:14
Aan: makerbase-mks/MKS-SGEN_L-V2
CC: Voxel3DPrinting; Author
Onderwerp: Re: [makerbase-mks/MKS-SGEN_L-V2] Encoder knob working wierd onLCD12864B with MKS SGEN L V2 (#26)

I want to know your endstop is normally open or normally closed and i think you should change endstop logic setting in configuration.h file

mks-viva, 2020-10-30:
Of course! If you has any problem when you using and you can feedback it to us.

Voxel3DPrinting, 2020-10-30:
Thank you so much!!
Van: mks-viva
Verzonden: vrijdag 30 oktober 2020 12:31
Aan: makerbase-mks/MKS-SGEN_L-V2
CC: Voxel3DPrinting; Author
Onderwerp: Re: [makerbase-mks/MKS-SGEN_L-V2] Encoder knob working wierd onLCD12864B with MKS SGEN L V2 (#26)

Of course! If you has any problem when you using and you can feedback it to us.

Voxel3DPrinting, 2020-10-30:
Hi,
I tried changing the SD from ONBOARD to LCD, but it still did not work, it now gave the error "SD INIT FAIL"..
However, inverting the endstop values did fix the homing issue.
So, the Homing and encoder problems have been fixed, but the SD problem still persists..

mks-viva, 2020-10-31:
> Hi,
> I tried changing the SD from ONBOARD to LCD, but it still did not work, it now gave the error "SD INIT FAIL"..
> However, inverting the endstop values did fix the homing issue.
> So, the Homing and encoder problems have been fixed, but the SD problem still persists..

Can you replace the short cable, from LCD to motherboard EXP1 EXP2, I think the cable is too long.

Voxel3DPrinting, 2020-10-31:
Nopee, still does not recognize the card at all.. not when it is set to LCD and not when it is set to ONBOARDmet vriendelijke groet, Sander Hesp

mks-viva, 2020-10-31:
> Nopee, still does not recognize the card at all.. not when it is set to LCD and not when it is set to ONBOARDmet vriendelijke groet, Sander Hesp

OK! I know what causes.
1. You need use TF card on the board and update firmware
2. You need remove motherboard's TF card and repower on, because LCD use SD_DET signl same as motherboard

Voxel3DPrinting, 2020-10-31:
It works!! Buuut, now I have a homing issue again... - _-IT says "homing failed, printer halted" but all endstops, stepper drivers and steppers work correctly and are connected right (endstop s->signal / v-> nothing / g->grond) What could this be? met vriendelijke groet, Sander Hesp

Voxel3DPrinting, 2020-11-01:
I also noticed that it is not  updating the firmware anymore, not via the LCD and not via the motherboard.
Steps I take:
build the firmware in platfirmIO, then locate the "firmware.bin" file, copy that to a 32gb TF card, formatted as FAT32, inserting that in the SD slot, putting power on the motherboard, and then pushing the resetbutton. I noticed the file on the TF card is not changing to "FIRMWARE. cur" as well.
What am I doing wrong?

Voxel3DPrinting, 2020-11-01:
I also noticed that the boot screen is all garbled up after I changed the bitmap in "dogm_bootscreen.h" file. What could cause this? I read it could be something to do with delays?
Video: https://youtu.be/l1pNhMdh52Y

mks-viva, 2020-11-02:
> It works!! Buuut, now I have a homing issue again... -_- After homing all axis, it says "homing failed, printer halted" but all endstops, stepper drivers and steppers work correctly and are connected right (endstop s->signal / v-> nothing / g->ground) If i home the axis individually it works fine, and do not get the error.  What could this be?

Can you provide a video？

mks-viva, 2020-11-16:
Has this problem been resolved？

Voxel3DPrinting, 2020-11-16:
Hey, thanks for coming back! 
The initial problem have been fixed, yes. Thankyou!
However I do have some issue's now regarding the heatbed; it just will not heat above 48 degrees. I am using an ender 3 24V heatbed, and the output on the cable is indeed 24.18 Volts. I cannot find where this is coming from. Maybe you can help me with this? I hope to hear from you soon.

Voxel3DPrinting, 2020-11-16:
Edit:
I measured the voltage on the motherboard tot 24,18 Volts. At the heatbed itself, it goes down to 23,7 Volts.
Could this make the difference?

mks-viva, 2020-11-16:
@Voxel3DPrinting 
If the heating cannot continue, it is largely a problem of the heating bed power. When heating, the voltage will drop because the hot bed is essentially a resistance

Voxel3DPrinting, 2020-11-16:
I noticed thanks! disable the PIDBEDTEMP in marlin and it is indeed working now!

The last problem I seem to have is the extruder making a clicking sound while retracting..
It is not present during extruding and I verified that I am using the correct Vref. I am using the tmc2100 stepper drivers with the e3d hemera hotend. What could this be?

mks-viva, 2020-11-20:
I think you can adjust the speed or distance of the extruder withdrawal when Cure slices.

doug-scott, 2020-11-28:
> I noticed thanks! disable the PIDBEDTEMP in marlin and it is indeed working now!
>
> The last problem I seem to have is the extruder making a clicking sound while retracting..
> It is not present during [...]

Easiest way to figure out the source of the sound is to feel the filament as it enters the extruder.  Since you are hearing this on retraction you will need to hold your fingers against the extruder while the filament is pushed out.  If you are feeling the filament stop moving when noise is made it is slipping in the extruder.  Depending on what extruder you have you may be able to increase the tension on the filament.  It is pretty odd to have it slip on retraction though.  How far is your print set to retract?  I have had it happen when the retraction is set too far and the filament was being compressed in the hotend, stopping it from going into the PTFE tube.  Note that this has also caused the filament to appear to be plugged in the hotend.

mks-viva, 2021-02-07:
I close this issues about encoder.

closed 2021-02-07 by mks-viva