makerbase-mks/MKS-SGEN_L-V2 #27: Board not installing Marlin

tatoap3, 2020-11-14:
Hi!

I just bought MKS SGen_L V2.0 and did all the steps that are on the guiding video makerbase put on github but when the board turns on it blinks the light and stabilize. Like in the video, but when I open the folder, it's still named "firmware.bin" and not "firmware.cur" like in the video. (the program compiles successfully) 

And I'd like some help to pin the board after connect the drivers. Which pin should I connect to use it without UART. I want to set VREF by the trimpot in each TMC2209.

Anyone knows what can I be doing wrong?

mks-viva, 2020-11-16:
Hi! 
If you use TMC2209 DIR/Step mode
1. You need use jumpers connect to motherboard's M0, M1; Driver IC power you can use 3.3V or 5V
[image]

If you use TMC2209 UART mode
1. Jumpers setting
[image]

2. If you use sensorless homing function, you need connect X-DIAG jumper and driver IC power connect to 3.3V
[image]

More information, you can refer to MKS SGEN_L V2 Wiki and link is: https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki

mks-viva, 2020-11-20:
Is this problem solved?

mks-viva, 2020-12-14:
Hi! I will close this issues temporarily.

closed 2020-12-14 by mks-viva

tatoap3, 2020-12-14:
Hi friend

Sorry to don't reply

I was trying to understand what is wrong, but I was without much time in
the past weeks

The printer I was trying to install didn't work with the board. I tried
with the makerbase display and with the one I had in the original kit, but
the motors don't move

I seted the same vref it was in the previous kit, did all the connections
right, checked over and over what could be wrong, but until now I don't
know what can be...

Now the printer is on one friend's home, if you can have a clue on what's
wrong and need any information please, let me know, so I ask him to take a
picture or check the information...

The original kit was mounted with an arduino mega and a ramps 1.4

Thanks for all the attention so far and sorry again for the time I took to
reply

Best regards

Otavio

mks-viva, 2020-12-14:
I suggest you follow my instructions to test it