makerbase-mks/MKS-SGEN_L-V2 #28: Mks sgenl v2.0 problem

AlexejLach, 2020-11-15:
mks sgen l v2.0 stopped working drivers burned out. if I put new ones, they are strong. lost connection with mks tft 35. through repieterhost everything works except for motors. As soon as I start the movement of the poevlyayets, the oshybka and the connection is given. please tell me what could be the problem

mks-viva, 2020-11-16:
Can you provide your motherboard's wiring picture or voides?

AlexejLach, 2020-11-16:
unfortunately I have already removed the board from the printer! put the old mks gen l v1.0

AlexejLach, 2020-11-16:
Can mks tft 35 work with mks gen l v1.0 board? I cannot get them to work

mks-viva, 2020-11-16:
> Can mks tft 35 work with mks gen l v1.0 board? I cannot get them to work

Of course! But you need set MKS TFT35's baudrate same as MKS Gen_L

AlexejLach, 2020-11-16:
in marlin should I change something

mks-viva, 2020-11-20:
MKS TFT35：
In mks_config.txt file : 
```
#baud rate (9600:1; 57600:2; 115200:3; 250000:4)
>cfg_baud_rate:4
```

MKS Gen_L(Marlin) in configuration file
```
/**
 * This setting determines the communication speed of the printer.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 250000
```

MKS TFT35 brudrate must same as MKS Gen_L

closed 2020-11-22 by AlexejLach