makerbase-mks/MKS-SGEN_L-V2 #29: Fan 1 and Fan 2

Kyosuke33, 2020-11-19:
Hello, i have a question, Fan 1 is for Layer Fan and Fan 2 is for Hotend Fan ? Or both are for Layer fan ?

Because i need to connect two layer fan, and at first i was thinking than both Fan 1 and Fan 2 are for layer Fan, but when i connect my two 5015 on Fan 1 and Fan 2, Fan 2 work for 10 seconde when i power on the printer, and after that it stop, and Fan 1 Work when i start layer fan at 100%.

mks-viva, 2020-11-20:
Hi! I am sorry, i don't quite understand what you think. Usually, people turn on FAN2 as a temperature control, for example, when the temperature is greater than 50 degrees Celsius, FAN2 will turn on...

Kyosuke33, 2020-11-20:
Ok, so Fan 2 is not a second "layer fan" who can be controled by the screen or Gcode. So if i have two fan to cool the print, i need to put both on the Fan 1.

mks-viva, 2020-11-23:
About FAN control:
FAN1 Open：M106 P0 S255   FAN1 Close：M106 P0 S0
FAN2 Open：M106 P1 S255   FAN2 Close：M106 P1 S0

Kyosuke33, 2020-11-23:
Ok, and in Marlin, where i can change the Fan 2 starting temperature ?

mks-viva, 2020-11-23:
In configuration_adv.h file
[image]

closed 2020-11-23 by Kyosuke33