makerbase-mks/MKS-SGEN_L-V2 #30: Using build MKS-SGEN_L-V2/Firmware/3D TOUCH/Marlin 2.0.6-3D TOUCH.rar

doug-scott, 2020-12-07:
I received the replacement motherboard the other day and am having issues.  I downloaded the "MKS-SGEN_L-V2/Firmware/3D TOUCH/Marlin 2.0.6-3D TOUCH.rar" build from here, took a quick look through the config, did not see anything unusual other than sdsupport being enabled so I disabled it(need to see motherboard sd card from pc) and also enabled the GCODE_CASE_INSENSITIVE that should be disabled by default, then I built it.  

Currently the motherboard has 4 x TMC2209 in UART mode, and the 3DTOUCH installed, connected to the z+ port, since the Z_MIN_PROBE_PIN is set to P1_24 in the CONFIGURATION.H file that is in the build archive.  I think I am confused about something, is it just me, and very likely it is, but, to me z_min is the same as z- and z_max is the same as z+.  Or is there no relationship between the names?  I see that P1_24 is in the z+ connection port, and P1_25 is in the z- connection port which is opposite to what I am thinking.

It updated the motherboard and I did the m502, and get the following from a m119 using Pronterface

SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED
z_probe: TRIGGERED
filament: TRIGGERED

When board is powered up the 3dtouch probe cycles twice, then sits there.  Sending commands to it results in nothing.  Using the M280 commands to test probe, and do not even get an error message.  I think the issue is the z_min appearing as TRIGGERED is likely the source, but I do not know where it is getting that from.  I have disconnected both connectors for the 3dtouch and it still comes back as TRIGGERED for the z_probe after a m119.  To me that should validate the wiring as not being the cause of that message.

Currently I can home x and y axis, and I can manually move the z-axis up and down, but cannot home z or complete a g28.  I have tried setting #define Z_MIN_PROBE_ENDSTOP_INVERTING  to be TRUE, the default was FALSE.  I cannot find where it is getting the triggered from, I was sure changing the inverting would fix it.  I have also tried changing the #define Z_MIN_ENDSTOP_INVERTING to TRUE from its default FALSE but it did not fix it.  I even tried setting both of those definitions to TRUE, but that also did not work.
I am also getting a different response today from the m119, today it returns

x_min: open
y_min: open
z_min: TRIGGERED
filament: TRIGGERED

Don't know where the z_probe response has gone.

mks-viva, 2020-12-09:
Can you provide your fimware to me?

doug-scott, 2020-12-10:
Firmware, or configuration files?  I have attached the configuration files.

[Configuration.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/5673894/Configuration.zip)

mks-viva, 2020-12-11:
I need firmware and i will build it and test it

doug-scott, 2020-12-11:
ok, it makes no sense to me how you can build the firmware from the firmware.bin file, but, you know what you want.  Here is the latest copy of the firmware.bin file.  I have also included the configuration files in case there is an issue with language translation and it is the build files you need.

[firmware.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/5679673/firmware.zip)

doug-scott, 2020-12-12:
I got the 3dprobe to work.  This probe was the one I was using with the motherboard that burnt.  I had ordered a spare 3dprobe a while ago and it got here today.  I stuck it in and how it works.  Seems the existing 3dprobe was also damaged when motherboard went out.  Now I have plugged the 12864 into the motherboard and it does not work.  It is defined in the configuration.h file, but when it comes on, it is just a bright blank screen.  No image of any kind, and pressing the reset button on it does not reset the board.  See attached picture.

[image]

mks-viva, 2020-12-14:
3D TOUCH‘s 3PIN interface need connect to P1.23 interface
3D TOUCH's 2PIN interface need connect to Z+ interface's S G
MKS has test it is will working

doug-scott, 2020-12-14:
The user manual on GitHub for the MKS SGEN_L v2.0 is not clear on which port to use for the 2 pin interface.  It actually shows to use either one, but, then it fails to mention what the PIN number should be for both.  It also uses the term Z_MIN to describe the connection which apparently is the Z- port.
The root problem on my setup was a defective 3dprobe device.  replaced the probe, started the printer up and it works now.  I am still getting the z_probe: TRIGGERED message when running the m119 command.  Tried both settings for the #define Z_MIN_PROBE_ENDSTOP_INVERTING and the #define Z_MIN_ENDSTOP_INVERTING settings.  I cannot get the z_probe to show OPEN.  I am not sure it matters, I have printed out a half dozen items since getting it up and running.
I have also moved to Marlin v2.0.7.2, I could not see the SD card in the motherboard using the "official" version.  I have commented out the #define SDSUPPORT to allow me to see the SD card.  
As I posted above the 12864 display stays on a blank bright screen.

The TFT35 v3.0 display I purchased to use instead of using 2 displays is not working correctly, and have been working with the vendor to sort it out.  The display will only display the touchscreen interface as long as I have a cable plugged into the EXT1 port with one end having the wiring installed wrong way around, the cable for the EXT2 socket must be a correctly wired cable.  With the cabling like that I can see and use the touchscreen, although it gives error about no printer, and the 12864 interface will show up on the screen, but, if I turn the encoder knob it will turn the extruder on when the encoder crosses a detent, then turn it off between detents.  It also keeps rebooting the display once you turn the know.  I can see the root screen, and the first page after that, but am unable to move between the options with the encoder wheel.  Still waiting for them to decide it is defective.

doug-scott, 2021-01-10:
Finally got everything working again.  It was mostly an understanding issue on my behalf.  I have a better understanding now, and am just awaiting a replacement 12864 display.

closed 2021-01-10 by doug-scott