makerbase-mks/MKS-SGEN_L-V2 #31: Compile problem with eeprom.

detroitpowa, 2020-12-07:
When i used the "define EEPROM_SETTINGS"  i all time got error with tmc2209 motor drivers, when the "#define X_DRIVER_TYPE  TMC2209" is not active i can compile it, but with motor drivers i don't. What is the problem?

Linking .pio/build/LPC1769/firmware.elf
/home/detro/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
collect2: error: ld returned 1 exit status
*** [.pio/build/LPC1769/firmware.elf] Error 1

mks-viva, 2020-12-08:
Hi! Can you provide your config file to me and i will check it?

detroitpowa, 2020-12-08:
Hi! Yes sure.

[asd.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/5662021/asd.zip)

mks-viva, 2020-12-09:
If you want to use EEPROM function, you need enable "#define EEPROM_SETTINGS" in Configuration.h file

detroitpowa, 2020-12-12:
Yes i know, but when i enabled it, i got firmware.elf error at the compiling end.

Linking .pio/build/LPC1769/firmware.elf
/home/detro/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
collect2: error: ld returned 1 exit status
*** [.pio/build/LPC1769/firmware.elf] Error 1

cizquierdof, 2020-12-12:
I got the same

detroitpowa, 2020-12-13:
I found this in an opened anoher issue folder.

"I think it is solved, the error appears when #define EEPROM SETTINGS is enabled and the screen is defined as #define TFT_480x320_SPI. That combination gives error, you have to choose to have eeprom settings or marlin UI.
#define SPI_GRAPHICAL_TFT and #define EEPROM SETTINGS does not give a problem"

That is a good one, it is fixed for me.

mks-viva, 2020-12-14:
@cizquierdof @detroitpowa 
So, i think your firmware can use this method

detroitpowa, 2020-12-14:
The printer is under building process. I not tested yet, but the compiling process is working well with eeprom. I hope this will be good. Tomorrow i will test it.

mks-viva, 2020-12-14:
@detroitpowa 
Ok, looking forward to your good news

detroitpowa, 2020-12-14:
I tried the newest looking //#define TFT_LVGL_UI_SPI  background mode, but when i compiled that, i have got some pin issues. That is complicated. Maybe in the future..

mks-viva, 2020-12-14:
MKS SGEN_L doesn't support TFT_LVGL_UI_SPI(MKS UI), only support Marlin Color UI and Marlin 12864 UI

detroitpowa, 2020-12-14:
Thanks for the info. :) On the web page where i bought the set (screen+board), the screen was the same like TFT_LVGL_UI_SPI, that's why i thought.

mks-viva, 2020-12-14:
We will add precautions, it can only be used in Marlin Color UI and Marlin 12864 UI when you use MKS SGEN_L motherboard.

detroitpowa, 2020-12-14:
I tried it with SPI_GRAPHICAL_TFT, but on the screen i got eeprom error. I switch back to TFT_480x320_SPI without eeprom.
There is no problem, maybe the display can store informations, actually i don't know.

mks-viva, 2020-12-15:
Maybe the marlin team did not consider using the SPI_GRAPHICAL_TFT style when doing this feature, we will continue to follow it up

ramboza, 2021-01-04:
so, finally, how to compile firmware to use EPROM with TMC2208 with UART?

the way described here is not working... 
https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/Drivers_MKS_TMC2225_2208

what can we disable to reduce memory usage?

mks-viva, 2021-01-05:
@ramboza 
Hi! This is only about how to use TMC2225/2208 uart mode settings, if you use EEPROM, you need enable "#define EEPROM_SETTINGS" in configuration.h file

ramboza, 2021-01-05:
> @ramboza
> Hi! This is only about how to use TMC2225/2208 uart mode settings, if you use EEPROM, you need enable "#define EEPROM_SETTINGS" in configuration.h file

Yes, I saw it in documentation, and you told it in previous replies, and I did it! 
just uncomment define for EEPROM_SETTINGS
And got error ! Have you tried replicate? 

How to replicate this issue: 
1 - download this https://github.com/makerbase-mks/MKS-SGEN_L-V2/tree/master/Firmware/Marlin-bugfix-2.0.x-MKS%20TS35%2BSGEN_L%20V2
2 - open it in Visual Studio Code
3 - uncomment #define EEPROM_SETTINGS
4 - compile 

Please, try it!

mks-viva, 2021-01-05:
@ramboza 
Hi! I has test it. You need change a parameter in pins_MKS_SGEN_L_V2.h file
#define TFT_BUFFER_SIZE                2400 change it to    #define TFT_BUFFER_SIZE                1200
[image]

ramboza, 2021-01-05:
Awesome!! it works now! 
Also 2000 as value is working! 

can you please update wiki page, it will save a time for someone else ;)

Thanks!

mks-viva, 2021-01-06:
@ramboza I will update pins_MKS_SGEN_L_V2.h file

closed 2021-03-06 by mks-viva