makerbase-mks/MKS-SGEN_L-V2 #33: Firmware crash dragging change value bar

cizquierdof, 2020-12-15:
SGEN L V2 with marlin 2 bugfix mks TS35 TMC2209 and COLOR Marlin UI.
Example with hotend temperature selection active if I drag de bar with the touch too quickly, screen goes black and it must reste or switch off the printer

mks-viva, 2020-12-16:
You can use the latest version marlin bugfix: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x
I will update MKS SGEN_L V2 path firmware

mks-viva, 2021-01-22:
Hi! If have not response, i will close this issues temporarily.

closed 2021-01-22 by mks-viva

tinyOtter, 2021-05-04:
Hello, Im having the same issue. when dragging the bar or turning the encoder knob ion TS35 there is a random reset.
compiled in following versions: 'Marlin-2.0.7.2', 'Marlin-2.0 Bugfix', 'Marlin-2.0.7', 'Marlin-2.0.6' and Marlin-2.x' [Basically the 2.() versions].
All of the versions have the same issue of random resets. at first i found i had a faulty encoder which was replaced quickly to a new one, BUT the issue is not gone.
Additionally, from the troubleshooting i moticed that only if 'Exp2' is connected the issue returns.

Im a total newbie in this, if need more info please ask.