makerbase-mks/MKS-SGEN_L-V2 #35: MKS ROBIN E3D z axis not respond after update the firmware

Xuculate, 2020-12-28:
Hi I'm a little desperate: 

Hi. I have recently installed the MKS Robin E3D board +2209.

The board worked fine with the Marlin 2.0.6.1 stock firmware. But I wanted to reverse the rotation of the extruder motor because I have installed a BMG. I have tried to put the firmware from here: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/ender3-3d-touch/firmware and now the z axis does not work, also I still do not know how to compile the Marlin configuration for reverse the extruder motor, etc. I am totally new to this Marlin thing. If you have a configuration for the CR10S to work fine it would be greatly appreciated. 

I have changed the controllers between them and they work fine, I have also changed the cables and the output of z does not give signal.  So I don't know if it's because of firmware or hardware.

I don't know what to do

Thank you!

mks-viva, 2020-12-29:
Are you using MKS Robin E3D V1.1 version motherboard? If yes, then the Z drive signal has been modified and you should download this firmware
Link：https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1
[image]

MKS Robin E3D/E3 V1.1 and V1.0 compare
https://github.com/MarlinFirmware/Marlin/pull/20216
[image]

mks-viva, 2020-12-29:
I will close it, you refer to here: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/58

closed 2020-12-29 by mks-viva