makerbase-mks/MKS-SGEN_L-V2 #36: How to change configuration to make a delta robot?

12343954, 2020-12-31:
I wanna make a delta robot via MKS-SGEN_L-V2 borad, TMC2225 and 57 steppers.
I burn the firmware `.\Firmware\Marlin-bugfix-2.0.x-MKS TS35+SGEN_L V2` into the board.

What can I do the next? thank you !

[image]

mks-viva, 2021-01-04:
Hi! You can refer to here: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config

12343954, 2021-01-04:
Thank you for reply.

I read that delta config, its robot like this, not my want.
[image]

mks-viva, 2021-01-05:
Ok!

mks-viva, 2021-01-06:
I will close this issues temporarily.

closed 2021-01-06 by mks-viva

12343954, 2021-01-06:
it was failed, when i meger the latest `marlin 2.0.8 + delta config 2.0.8 + MKS-SGEN_L-V2 + TS35 v2+ TMC2225`.
i selected `step/dir` mode, default driver type(No uncomment).

stepper motors can't move smoothly, they vibrate very strong.