makerbase-mks/MKS-SGEN_L-V2 #37: ERROR: TMC Connection Error ! (TMC2225 + MKS SGEN L V2.0, Step/DIR mode)

12343954, 2021-01-01:
1. MKS SGEN L V2.0, Step/DIR mode, 3 tmc2225 and 3 stepper servoes

2. following the tutourial: https://www.youtube.com/watch?v=6RcrgmNvyeA&ab_channel=MakerbaseTeam

3. flash the `Firmware\Marlin-bugfix-2.0.x-MKS TS35+SGEN_L V2`

4. after reboot, LCD show `TMC Connection Error`, what does this mean?

```
>>> M122
SENDING:M122
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW

```

mks-viva, 2021-01-03:
MKS SGEN_L V2 use MKS TMC2225 uart mode settings
1. MKS SGEN_L V2 motherboard jumpers settings
[image]
2. Marlin firmware configuration.h file set driver type to TMC2208（only package difference）
[image]
3. You need use DC12/24V power on

12343954, 2021-01-04:
Thank you, follow your step, it's not work, still  `TMC Connection Error`

I use default configuration to continue to develop.

mks-viva, 2021-01-04:
Then, you can directly use standard mode instead of uart mode

12343954, 2021-01-04:
yeap, but I don't know the difference of 3 modes `uarp`, `spi`, `step/dir`.

mks-viva, 2021-01-06:
Uart Mode: the drivers parameters can be change or config online, for example, MKS TMC2209,MKS TMC2225,MKS TMC2208 support uart mode
SPI Mode:  the drivers parameters can be change or config online, for example, MKS TMC2130, but we recommend users to use MKS TMC2209 instead of TMC2130
STEP/DIR Mode: The parameters config by hardware, for example, the drivers current

12343954, 2021-01-06:
oh, i see, thank you so much !

mks-viva, 2021-01-06:
So, i will close this issues temporarily.

closed 2021-01-06 by mks-viva