makerbase-mks/MKS-SGEN_L-V2 #38: Using second z-axis port on motherboard

doug-scott, 2021-01-02:
I am trying to correct an issue I have with running the z-axis stepper motors through one port.  This motherboard has a second z-axis connection that I am hoping will use its own driver.  I know that if I use the second z-axis port that I will no longer be able to use the second extruder.  Is there some documentation regarding how to configure Marlin to use the second z-axis connection?  The issue I am trying to address is if I try to move the z-axis more than about 20mm at a time the two motors will create a vibration in the bed that will cause one side to miss some of the steps, causing the bed to become out of sync.  I know I can just slow the feedrate down, but it would have to run at such a slow rate that any large movements just take too long.  Will running the second z-axis connection fix this issue?

mks-viva, 2021-01-03:
Hi! 
1. About Dual Z-axis set, you can set it by configuration_adv.h file
[image]
2. About motors will create a vibration, i think the current too small, you can set it by potentiometer knob

doug-scott, 2021-01-11:
Finally getting to this.  I only need to change the line you have outlined to "#define NUM_Z_STEPPER_DRIVERS 2"?  No other changes?  

I am using the 3DTOUCH probe for z-axis endstop, am I fine to leave it as is?  I am also going to be using TMC2209 on all 5 ports, and all 5 are set to UART mode.

mks-viva, 2021-01-11:
Of course, you can understand

TempestWales, 2021-01-12:
> Hi!
> 
> 1. About Dual Z-axis set, you can set it by configuration_adv.h file
>    [image]
> 2. About motors will create a vibration, i think the current too small, you can set it by potentiometer knob

With regards to the motor vibration, I'm unable to eliminate the z-axis motors missing steps and crashing into the platform when performing quick moves in Dual-Z Mode.

I currently have MKS TMC2209 v2 in Uart + Sensorless mode. If I disconnect one of the motors the other will run file. I've also increased the vref through UART but still unable to eliminate the grinding/missing steps.

Is there anything else I can do to get both working?

doug-scott, 2021-01-13:
> With regards to the motor vibration, I'm unable to eliminate the z-axis motors missing steps and crashing into the platform when performing quick moves in Dual-Z Mode. Slower or manual moves are perfectly fine.
> 
> I currently have MKS TMC2209 v2 in Uart + Sensorless mode. If I disconnect one of the motors the other will run file. I've also increased the vref through UART but still unable to eliminate the grinding/missing steps.
> 
> Is there anything else I can do to get both working?

Thats when I get the vibration/noise/one side basically stops moving and that just twists the bed and stops everything.  If I only move in 10mm pieces I can have it run as fast as I can set it, but, if the bed is going to move more than 10mm in one step it will fail everytime if it is not set to a slow number.  It may be the setup I have for the z-axis.  Mine uses the 8mm jack screw and has 2 8mm smooth rods per side with sleeve type bearing.  I have not removed them yet to see if that actually have ball bearings in them or are just a plastic sleeve.  I would imagine if I could just stiffen up the setup this would fix the issue.  Been looking for a setup that will allow me to move to linear rail, with a more robust bed mount.  When printing it is never as issue, it only shows up when setting up the print job to start printing.

TempestWales, 2021-01-29:
> Thats when I get the vibration/noise/one side basically stops moving and that just twists the bed and stops everything. If I only move in 10mm pieces I can have it run as fast as I can set it, but, [...]

I've tried running both or either one with a 1 screw and none, The issues occurs so long as two motors are connected, even with no screws attached, I'm wondering if its a config/UART issue. Really need some MKS input here.

doug-scott, 2021-01-30:
> 
> I've tried running both or either one with a 1 screw and none, The issues occurs so long as two motors are connected, even with no screws attached, I'm wondering if its a config/UART issue. Really need some MKS input here.

If you are only driving one side of the bed and essentially counting on the bed/frame to be solid enough to not distort or sag when being lifted by just the one side that makes it sound like when both sides are being driven, they are fighting amongst themselves for control of raising the bed.  I think I am going to try to make a bed frame from 2020 or 2040 extruded aluminum connected to a wheeled sled.  The wheeled sled makes it very difficult for the frame to go out of alignment, plus it removes any chance of out of square frame.

oldman4U, 2021-02-13:
Hi.

It would be great if you could close your ticket: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1586

Thank you

rth630, 2021-05-08:
The two Z sockets are electrically connected. One driver for both. Config Driver = 1. If you want a driver for each motor, the second has to be in the second extruter connection and config Driver = 2. If the printer has a connecting strap like the Artillery Sidewinder, remove it.