makerbase-mks/MKS-SGEN_L-V2 #39: PLEASE upgrade marlin to the latest version `CONFIGURATION_H_VERSION 020008`, I can not meger the latest config to your version `020006`!

12343954, 2021-01-05:
PLEASE upgrade marlin to the latest version as your "https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/Marlin_firmware/README" suggestion.

The marlin example config version is `020008`, and there are lots of differences between the latest version and yours.

I can not meger the latest config to version `020006`! Please help, thank you !

[image]

[image]

mks-viva, 2021-01-05:
Hi! You want to update it to the latest version marlin?

12343954, 2021-01-05:
en, for now, yes, because I know nothing about marlin, I'm a novice.

I don't know how are these pins defined, and how to use stepper motor directly. I have a kinematic algorithm.

mks-viva, 2021-01-06:
The latest version marlin firmware link: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

mks-viva, 2021-01-07:
Hi! I will close this issues temporarily.

closed 2021-01-07 by mks-viva

12343954, 2021-01-09:
I suggest you upgrade the source code, because you are a hardware supplier. upgrade together with marlin is good for your market.