makerbase-mks/MKS-SGEN_L-V2 #40: MKS MINI12864 V2.0

Hiroshi-OssanNo2, 2021-01-08:
MKS SGEN_L_V2
MKS MINI12864 V2.0

bugfix-2.0.x&Marlin-2.0.7.2

i used a simple config to check the problem
display will show "SD Init Fail"
can't access the SD card
Is this a Marllin problem?

Configration.h

#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define MOTHERBOARD BOARD_MKS_SGEN_L_V2

#define SDSUPPORT
#define SPEAKER
#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 2
#define LCD_FEEDBACK_FREQUENCY_HZ 5000
#define MKS_MINI_12864

Configration_adv.h

#define SDCARD_CONNECTION LCD

mks-viva, 2021-01-09:
Hi! I will test it

Hiroshi-OssanNo2, 2021-01-13:
i connect MKS MINI to other board
reslut is all ok

broken MKS_GEN_L_V2 (I bought a new one. .. ..
or
marlin bug

mks-viva, 2021-01-14:
Hi! I have test the latest version marlin bugfix 2.0.x
1. Configuration.h file set it
A. define board is MKS_SGEN_L_V2
B. Enable MKS_MINI_12864
C. Enable #define SDSUPPORT

2. pins_MKS_SGEN_L_V2.h file set
```
#ifndef SDCARD_CONNECTION
  #define SDCARD_CONNECTION              LCD
#endif
```

3、You need use motherboard‘s tf card update firmware and done you need remove motherboard's TF card, you can insert SD to MKS MINI12864

mks-viva, 2021-01-14:
My test firmware bin file
[firmware.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/5813151/firmware.zip)

Hiroshi-OssanNo2, 2021-01-15:
thanks mks-viva
tested config&test firmware
result is the worst
very unfortunately, my mainboard seems to be broken
i just bought it

mks-viva, 2021-01-16:
Oh! It sounds like it’s not good news.

closed 2021-01-22 by mks-viva