makerbase-mks/MKS-SGEN_L-V2 #41: UART Mode not working even on 24VDC

arnaud21270, 2021-01-09:
Hello, I have TMC2225 Drivers, I put the jumper regarding the tutorial and configure marlin bugfix 2.0.x to use uart mode on 4 drivers (X, Y, Z, E0), I switch on 24vdc but when I test the UART state I got this : 
>>> m122
SENDING:M122
axis:pwm_scale/curr_scale|flags|warncount
		X	Y	Z	E
Enabled		false	false	false	false
Set current	500	500	500	500
RMS current	883	883	883	883
MAX current	1245	1245	1245	1245
Run current	15/31	15/31	15/31	15/31
Hold current	7/31	7/31	7/31	7/31
CS actual	0/31	0/31	0/31	0/31
PWM scale
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	false	false	false	false
msteps		256	256	256	256
interp		false	false	false	false
tstep		0	0	0	0
PWM thresh.	1581	1581	10541	4534
[mm/s]		100	100	3	30
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
pwm scale sum	0	0	0	0
pwm scale auto	0	0	0	0
pwm offset auto	0	0	0	0
pwm grad auto	0	0	0	0
off time	0	0	0	0
blank time	16	16	16	16
hysteresis
 -end		-3	-3	-3	-3
 -start		1	1	1	1
Stallguard thrs
uStep count	0	0	0	0
DRVSTATUS	X	Y	Z	E
sg_result
stst		*	*	*	*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW

any idea?
Thanks in advance.

mks-viva, 2021-01-11:
Hi! Are you set driver type in Configuration.h file
[image]

arnaud21270, 2021-01-11:
Hello, yes I have done it like in your example as it says that for TMC2225 we must adjust TMC2208.

mks-viva, 2021-01-14:
Use uart mode:
1. Jumpers set(X Y Z E0 use uart mode) and you need use MKS TMC2225 connect to MKS SGEN_L V2 motherboard
[image]
2. Marlin configuration.h file set
[image]
3. You need use DC12-24V power on。

arnaud21270, 2021-01-22:
It is exactly what I've done, but when I send M122 I get the message I've post on the first message. Subject is not closed! thx

arnaud21270, 2021-01-25:
ok it is still open but what next?

mks-viva, 2021-01-25:
Provide your firmware to me and i will use it test.

mks-viva, 2021-01-26:
Oh No! I always thought that you are using MKS TMC2225. If you are using MKS TMC2225, then you can use Uart mode directly by plugging it in, because MKS TMC2225 mode supports Uart mode. However, if you are currently using the TMC2225 module of another manufacturer, you may need to set the TMC2225 module to Uart mode according to the manufacturer's instructions, and may need welding or flying wiring.

arnaud21270, 2021-01-26:
ok thanks, I just see the issue with the stepper. The pins doesn't match with your version.

mks-viva, 2021-01-26:
Yes, if you use MKS TMC2225 directly, then no welding is required

arnaud21270, 2021-01-26:
I will check if I can try to wire the drivers I have to use uart mode.
Thanks for your help, you can close the subject.

mks-viva, 2021-01-26:
Ok! I will close this issues temporarily.

closed 2021-01-26 by mks-viva