makerbase-mks/MKS-SGEN_L-V2 #44: Babystepping using TFT 3.5 causes hang

V3nn3tj3, 2021-02-03:
I recently upgraded my old Marlin to the Sgen 2.
When using babystepping in the print menu on the TFT3.5 the printer stalls or hangs for a little while. It does not keep printing.

This doesn't work as the filament keeps running out.

Am i doing something wrong that could cause the printer to hang?

mks-viva, 2021-02-07:
Hi，I want to know，if you have not printing.  The baby stepping function is ok?

V3nn3tj3, 2021-02-08:
I can't see the babystepping when not printing.
I mapped the MORE buttons to "M290 Z", and that is working. Babystepping does work, it is just when opening the babystepping provided on the TFT3.5 that the SGEN board crashes.

mks-viva, 2021-02-08:
So，we need check Marlin firmware settings

V3nn3tj3, 2021-02-08:
Marlin firmware is ok, babystepping works with custom buttons. Only not when using separate babystepping menu from the TFT screen.

Magiquall, 2021-03-06:
Hello,

Did you find a solution ? I am in the same case and it does not work as soon as we open the babystepp view

Magiquall, 2021-03-20:
Hello, in fact it's not just the BabyStepp that crashes the card, it's as soon as I make a print then in the options menu, and immediately after entering into a menu babystepp, fan or other, the printer crashes. We really feel alone with these issues, Makerbase, thank you for helping us.