makerbase-mks/MKS-SGEN_L-V2 #45: Extruder fan not start automatically on my Tarantula Pro Orange version

ILGROSSOFA, 2021-02-06:
Dear Sirs,
I changed mobo and display on my TT PRO, installing MKS SGEN_LV2.0 with TS35 display, 3D touch and TMC2208 V2.0 UART.
Everything is running fine but I have a  problem with the one of the 3 fans installed on the hotend.
2 lateral fans are for layer cooling and they are running fine, but the 3rd central fan which is for extruder cooling doesn't start when extruder heats up.
But if I go through display menu and I increase the percentage of fan 2 (which is at 0%) so it starts and then keeps running.
About wiring I copied the connections as for the previous mobo which was MKS SGEN_L V1.0 so maybe I have to do something in the firmware but I don't know what and where.
Please help me, thank you very much!
GROSSO

ILGROSSOFA, 2021-02-07:
SOLVED!
In configuration.adv.h file:
#define E0_AUTO_FAN_PIN P2_06

mks-viva, 2021-02-07:
So，I will lose this issues temporalily

closed 2021-02-07 by mks-viva