makerbase-mks/MKS-SGEN_L-V2 #46: Firmware modifications for dual Z axis upgrade

ILGROSSOFA, 2021-02-07:
I would like to add a dual Z axis stepper motor and driver but I'm not sure how to proceed.
First of all where should I connect the motor? Z2 or E1 port?
If I connect motor to Z2 port still I have to add the stepper driver in the 5th place?
Which modification should I do in firmware?
So I miss a lot of info, please help me, thank you
GROSSO

mks-viva, 2021-02-07:
If you use Z2 interface，you need not set firmware

TempestWales, 2021-02-08:
> If you use Z2 interface，you need not set firmware

When I connect the second motor to the Z2 port, both of them will skip steps when moving the Z-Axis on my pre-gcode (See below) movement. Moving the Z-Axis manually is fine as well as slow movements. This issue occurs even if the second motor isn't connected to a screw, and goes away when either motor is used individually.

> G90 ; switch back to absolute mode
> G1 Z30.0 F9000 ;move the gentry up 30mm
> G92 E0 ;zero the extruded length again

ILGROSSOFA, 2021-02-19:
> If you use Z2 interface，you need not set firmware

Hi, 
I want to use the E1 port and a dedicated driver for Z2 axis, in order to be able to use auto Z align.

mks-viva, 2021-02-24:
> > If you use Z2 interface，you need not set firmware
> 
> Hi,
> I don't want to share one driver for dual Z, I want to use a dedicated driver for Z2 axis, in order to be able to use auto Z align.

E1 as Z2
[image]

ILGROSSOFA, 2021-03-03:
> > > If you use Z2 interface，you need not set firmware
> > 
> > 
> > Hi,
> > I don't want to share one driver for dual Z, I want to use a dedicated driver for Z2 axis, in order to be able to use auto Z align.
> 
> E1 as Z2
> [image]

Hello Sirs,
 if I understand well I have to use the E1 port and not Z2 port to be able to use its own stepper driver, right?
And then I indicate #define NUM_Z_STEPPER_DRIVERS 2.

In configuration.h file what is the correct configuration to enable Z2 on E1 port?
Thank you.
Best regards
GROSSO

mks-viva, 2021-03-05:
@ILGROSSOFA If you want to E2 as Z2, so, you need change it on configuration_adv.h file: 
#define NUM_Z_STEPPER_DRIVERS 2
[image]

closed 2021-03-06 by mks-viva