makerbase-mks/MKS-SGEN_L-V2 #49: MKS LCD12864B only backlight, sound and reset working

ToM4s85, 2021-03-04:
Hello, I've recently bought the MKS SGEN L v2 to replace my old board and bought the LCD with it. I've tried flashing the premade firmware but it didn't work. I also tried making my own Marlin on the release version and on the Nightly build but neither of them worked. The display's buzzer does work when I turn the board on, the reset button also works. But the display is only turning on the backlight and doesn't display anything. I also can't access the SD card that's in the display when I connect through USB to my computer. I looked through here and the Marlin Github, but didn't find any working solutions. Any help is appreciated.

Voxel3DPrinting, 2021-03-04:
did you uncomment the right LCD type in the configuration.h file?

ToM4s85, 2021-03-04:
Yes, I did, I tried it with multiple display configs that were suggested. Tried MKS_LCD12864 and also tried with MKS_MINI_12864 because they have the same pinout. I also tried the premade firmware from this Github that didn't work either.

mks-viva, 2021-03-05:
@ToM4s85 Which version of firmware you use?

ToM4s85, 2021-03-05:
I used the newest Marlin 2.0 release and also tried nightly from February 27th. I tried the premade firmware TMC2209 + LCD12864.bin from the SGEN_L V2-bin folder. None have worked so far.

mks-viva, 2021-03-06:
I has test it ok
1. Set borad as BOARD_MKS_SGEN_L_V2
2. Enable MKS_LCD12864
3. Enable SDSUPPORT
My test firmware
[Marlin-bugfix-2.0.x+SGEN_L V2+MKS LCD12864B.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/6095038/Marlin-bugfix-2.0.x%2BSGEN_L.V2%2BMKS.LCD12864B.zip)
[firmware bin file.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/6095040/firmware.bin.file.zip)

ToM4s85, 2021-03-07:
OK, I have replicated your setup of the board and it works. But there is a problem, I have to have my drivers in Dir/Step mode for it to boot the first time, after that I can turn it off, put it to Uart mode and the board boots, but I left it disconnected overnight and in the morning I had to repeat the whole driver mode change for it to boot. Do you have any idea why there can be a problem with booting to Uart driver mode after some time without electricity, but when I just disconnect and connect a few minutes later it works fine? I am using MKS_TMC2209 drivers.

mks-viva, 2021-03-08:
If you use TMC2209 Uart mode, you need connect DC12-24 power, otherwise, a TMC error will occur

ToM4s85, 2021-03-08:
Ok, in the meantime I have fixed the issue. It may have been caused by using one driver that I did not define in the firmware. It seems to work without error now. I was booting with a power supply so I don't think that the TMC error was the cause of failed boot. It seems to work without a problem now, thanks for your help with this issue.

closed 2021-03-08 by ToM4s85