makerbase-mks/MKS-SGEN_L-V2 #50: How do you use TMC2209 with MKS SGEN L V2??

Voxel3DPrinting, 2021-03-04:
Hey,
A while ago I bought the MKS Sgen L V2 motherboard and used it with the tmc2100 stepper drivers. However, after I blew up the drivers while trying to put them in spreadcycle mode..
Because of this, I bought the MKS tmc2209 drivers and I wonder if there is any video/document available on how to use these drivers and enable the stallguard (sensorless homing) in this combo. (MKS Sgen L V2 and MKS tmc2209)
I hope to hear from you.
Thanks

Voxel3DPrinting, 2021-03-04:
to clarify; I plan to use these drivers on the X, Y and Z axis

mks-viva, 2021-03-05:
Hi! You can refer to here : https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/Drivers_MKS_TMC2209

closed 2021-03-06 by mks-viva