makerbase-mks/MKS-SGEN_L-V2 #51: MKS-SGEN_L-V2 + MKS TFT35

Magiquall, 2021-03-06:
Hello,
I had a babystepp problem with the Marlin 2.0.6 version available on Makerbase. (MKS SGEN_L V2 + TFT35)
So i used the latest version of marlin bugfix 020008,
but I have a problem with the screen, because I must surely configure it incorrectly.
In your 2.0.6 you declare
#define TFT_480x320_SPI
But it does not exist in the latest marlin. I managed to compile by activating TFTs with the same resolution, but now here is my concern:

My problem now:
Everything works fine manually, but when I start printing the plate is heating up, but the temperature indication does not move. yet arrived at 50 ° C it stops heating, but on the screen it has not moved and it does not go further. Printing does not start and does not heat the nozzle.

I specify:
Manually when I go to preheat, the temperature is well indicated and when the plate heats up, the displayed temperature rises well. But not when I run the print.

mks-viva, 2021-03-06:
Hi! USe MKS TFT35 or MKS TS35?

Magiquall, 2021-03-06:
MKS TFT35

https://fr.aliexpress.com/item/32891320510.html?spm=a2g0s.9042311.0.0.27426c37Ov58IR

thank you, Excuse me for my limited english

mks-viva, 2021-03-06:
MKS TFT35 have not support marlin UI, so, you only use serial function
1. Set #define SERIAL_PORT 0 in configuration.h file
2. Connect MKS TFT35 by AUX-1
3. You can refer to Wiki : https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/Display_MKS_TFT35

Magiquall, 2021-03-06:
OK, I just understood my error, in your firmware (available on your Github) I made a mistake:
Marlin-bugfix-2.0.x-MKS TS35 + SGEN_L V2
This is for the TS35 and not the TFT35, excuse me.
So for the TFT35 if I understood correctly, we have nothing to do in the Marlin, apart from opening communication via the COM port.

mks-viva, 2021-03-06:
Yes it is

closed 2021-03-06 by mks-viva