makerbase-mks/MKS-SGEN_L-V2 #52: No respondes to M500, M501 and M502

Haschi0, 2021-03-07:
Hi guys,

This is my first ever post on Github and I hope I'm reasonably correct here. To be polite, I would like to introduce myself briefly. I live in Germany and with my early 50s I certainly no longer belong to the classic clientele of 3D printer operators, I think. ;-) 
My printer is a JG Aurora A5, about 3 years old, with various modifications. I am an electronics engineer.
 
I recently bought a new MKS SGEN L V2.0 and built a firmware with Marlin bugfix-2.0.x and PlatformIO.
I have copied the settings line by line from my previous configuration.h and configuartion_adv.h from Marlin 1.1.8. worked on a MKS GEN L 1.0 For now, I have left the new settings in their default settings. 

I have also added the:
#define EEPROM_SETTINGS // Persistent storage with M500 and M501
//#define DISABLE_M503 // Saves ~2700 bytes of PROGMEM. Disable for release!
#define EEPROM_CHITCHAT // Give feedback on EEPROM commands. Disable to save PROGMEM.

When I try to perform mesh bed leveling with G29 S1, it appears to work. However, when I try to save the data to the EEPROM with M500, I get no response. (xxx bytes written, CRC xxxx) and when I query the data with G29 S0, I get the error message: Failed to enable Bed Leveling. The same with M420.

Since I could not save the leveling data with M500 before, I tried an M502 reset with M500 save. Both failed to return a response, but an M503 generates a printout of all values (my compiled base values).

Does anyone have an idea for me where to look for. Auntie Google couldn't really help me so far.

Thank you very much for your efforts.

mks-viva, 2021-03-08:
Hi! Can you provide your firmware to me?

Haschi0, 2021-03-08:
What exactly you would like to have? The two configuration files or the whole source or really the firmware.bin?

Haschi0, 2021-03-08:
I think you want to test my firmware on another HW. Okay, here is my firmware.bin as a zip file and also my two configuration files

[firmware.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/6103173/firmware.zip)
[Marlin Configuration.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/6103195/Marlin.Configuration.zip)

mks-viva, 2021-03-09:
I think you need enable "EEPROM_AUTO_INIT", like this
[image]

Haschi0, 2021-03-09:
I have activated the "EEPROM_AUTO_INIT" but I still don't get a response on M500. 

So I looked at Marlin's source code and found something that might explain why I'm not getting a response:

- The M500 command (in M500-M504.cpp) calls settings.save(). 
- This is in the file settings.cpp beginning from line 604.
- My previously known response to M500 (on my previous board MKS GEN L V1.0 with Marlin 1.1.8) was allways a "Setting Stored (xxx bytes; crc xxx)". 
- The output of this response is in line 1410.
- This is enclosed with a "DEBUG_ECHO_STEART()" and a "DEBUG_ECHOLNPAIR(...). 
- Both functions are extended by the debug_out.h file with a "NOOP".
- This in turn is due to the fact that "DEBUG_OUT" is not defined.
- Which is set by: #define DEBUG_OUT ENABLED(DEBUG_LEVELING_FEATURE) in settings.cpp in line 565.

In short, I would have to enable the "DEBUG_LEVELING_FEATURE" in Confoguration.cpp to get a confirmation response on M500.

After a test by an M603 with setting of length filament load and unload and a restart of the board, I see after M503 that the changed values were stored in the EEPROM. Without having to call the M500 explicitly.
It seems to me that the differences between Marlin 1.x and 2.x are greater than I've  expected.

Many thanks for your help, MKS-viva. 
Now I just have to find out why the Mesh Bed Leveling does not work. 
But that is another story.

closed 2021-03-12 by Haschi0