makerbase-mks/MKS-SGEN_L-V2 #53: SD init fail issue in 2004 lcd

Cheon-dev, 2021-03-07:
Hello, my mks sgen l v2 board can't connect to the sd card
I am using lcd2004 and I modified the firmware by looking at other issues before, but it is not recognized.

configuration.h ->  
#define SDSUPPORT
#define REPRAP_DISCOUNT_SMART_CONTROLLER

configuration_adv.h ->
#define SDCARD_CONNECTION LCD
(This is the same result as enable/disable)

pins_MKS_SGEN_L_V2.h->
#define SDCARD_CONNECTION LCD

I've replaced all the short cables and long cables. But still sd init fail.

mks-viva, 2021-03-08:
Hi! Can you provide your firmware to me and i will test it?

Pac333, 2021-03-08:
This is the firmware file I uploaded
[firm.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/6098557/firm.zip)

mks-viva, 2021-03-08:
You need remove board's TF card, because LCD's SD and board‘s TF card use SD_DET signl is the same

Pac333, 2021-03-08:
Of course, I also referred to the previous issue. 
but sd init fail continuously

mks-viva, 2021-03-08:
This issues and i test it is ok

Pac333, 2021-03-08:
Is my firmware ok..?
I couldn't sleep all night because of this problem.
The only way left is to replace the lcd with a new one...🥲

mks-viva, 2021-03-08:
Yes, i test it is ok, you need remove Board's Tf card

Pac333, 2021-03-08:
After removing the board tf card, eeprom initialization proceeds, and the sd card of the lcd is recognized!

I have other problems, can I solve them?
1. A long beep sounds for the printer boot and the power is turned on.
2. The extruder fan (not a hotend fan) works the first time the power is turned on, but stops working when entering the main menu -> It does not work even during printing.

mks-viva, 2021-03-10:
About your other problems
1. This is normal 
2. You need to define its function by yourself, for example, it will be turned on when the heating head temperature reaches

Pac333, 2021-03-11:
The buzzer was capped to reduce noise
The extruder fan was connected to the E1 terminal, not the regular power terminal.
All problems have been resolved, thank you :)

closed 2021-03-12 by Pac333