makerbase-mks/MKS-SGEN_L-V2 #54: Marlin config probs for MKS_PWC

ILGROSSOFA, 2021-03-12:
Dear Sirs,
you write to modify Configuration.h as follows：

Enable #define PSU_CONTROL
Set #define PSU_ACTIVE_HIGH true

But the line **Set #define PSU_ACTIVE_HIGH true** doesn't exist in Marlin-bugfix-2.0.x.

There is written as follows:
**#define PSU_ACTIVE_STATE LOW      // Set 'LOW' for ATX, 'HIGH' for X-Box**

What should I do?

Thanks in davnce,
Best regards
GROSSO

mks-viva, 2021-03-16:
Which version of firmware do you use?

mks-viva, 2021-03-25:
The latest version marlin firmware, you can set  #define PSU_ACTIVE_STATE HIGH and add it to pins_MKS_SGEN_L_V2.h file
```
#if ENABLED(PSU_CONTROL)
#ifndef PS_ON_PIN
  #define PS_ON_PIN         P1_23    //PW_OFF, you can change it to other pin
#endif
#ifndef KILL_PIN
  #define KILL_PIN          P1_28    //PW_DET, you can change it to other pin
#endif
#define KILL_PIN_INVERTING  true     //true : HIGH level trigger
#endif
```

closed 2021-04-13 by mks-viva