makerbase-mks/MKS-SGEN_L-V2 #56: The temperature is not updated during heating.

ChristopheSV, 2021-03-22:
Hello,

I don't understand I no longer have real-time temperature feedback when launching a print.

The temperature of the bed is displayed when I enter a menu or when I run a print, but it does not change, although the bed is heating up well. The heating stops when it reaches the set point (50 ° C), but on the MKS TFT35 screen it always displays the original value at 22 ° C, and therefore the printing is always waiting.
If I stop printing and go to the preheat menu, the temperature shows 50 ° C, but if I let it cool down, there is still 50 ° C, if I exit the menu, and enter the menu again. menu, this displays the new value to me.

I specify that this function worked I was able to print some parts, but it does not work any more. I tried to flash the motherboard again, then the screen, but nothing changed.

Where does it come from? Because the value does arrive on the map, but it is not updated.

mks-viva, 2021-04-14:
Can you update the TFT35 firmware to the latest version?

ChristopheSV, 2021-04-14:
Hello,
I had already put the latest version in effect.
But you can close the topic, because I have unplugged everything since the time, and I went to Bigtreeteck and I have no more worries. You have to react more quickly when you have problems, you really feel neglected at Makerbase. A lot of people I have contacted on the forums have abandoned Makerbase because of this. It's a shame, because you had an excellent reputation which seems very compromised today. I really hope you will find yourself in the hearts of the "makers", but for the moment I have given up.
Good continuation.

mks-viva, 2021-04-15:
Maybe we need improvement in many areas, thank you!

mks-viva, 2021-04-15:
But about this issue, can you provide me with a video of the situation at work? I will check it.

closed 2021-04-15 by mks-viva

ChristopheSV, 2021-04-15:
Hello,

Thank you for your support.
I'm going to reinstall the card for you and make a video, but I'll do it tomorrow, because I'll be on vacation. Currently, I am at work, so I don't have a lot of time.
But unless there's something unforeseen, I'll give you the video tomorrow.

Thank you,

ChristopheSV, 2021-04-16:
Hello,
You can find my video, sorry for my poor english.
So as you can see, the temperature in the print view is not moving at all. So it doesn't go any further than the bed heater control.

In the video I show you that if we go to the other temperature control menu it's perfect, but in the print view, it takes the temperature when entering and then it does not move.

https://drive.google.com/file/d/11aVUS6Sjy2Dfp9OgkBzqZ4JanQ2xiC3Z/view?usp=sharing

mks-viva, 2021-04-17:
Hi! I has test MKS TFT35 the latest version firmware（1.0.8） and heater bed and temp has update. Mayb be you need provide your SGEN_L V2's firmware and your TFT35's firmware to me and i need test it.
[image]

https://user-images.githubusercontent.com/16629551/115105083-3697f480-9f8f-11eb-8e3f-4cab1d758abf.mp4