makerbase-mks/MKS-SGEN_L-V2 #57: Connect MKS-SGEN_L-V2 and RPi3 via RX TX

rom-ka1, 2021-04-11:
Hello! 
No connection MKS-SGEN_L-V2 and RPi3 via RX TX, using port AUX-1.

3-pin connector cable
MKS-SGEN_L-V2      **GND(black) TX(blue) RX(red)**      ->      RPi3       **GND(black) TX(red) RX(blue)**

 **/boot/config.txt**
add 
`dtoverlay=pi3-miniuart-bt`

**/boot/cmdline.txt**
delete
`console=serial0,115200`

**printer.cfg**
```
[mcu]
serial = /dev/ttyAMA0
restart_method = command
```

**make menuconfig**
```
[*] Enable extra low-level configuration options
    Micro-controller Architecture (LPC176x (Smoothieboard))  --->
    Processor model (lpc1769 (120 MHz))  --->
[*] Target board uses Smoothieware bootloader
    Communication interface (Serial)  --->
(250000) Baud rate for serial port (NEW)
[ ] Specify a custom step pulse duration (NEW)
()  GPIO pins to set at micro-controller startup (NEW)
```

Using a USB cable, there is a connection.
[klippy.log](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/6291223/klippy.log)

rom-ka1, 2021-04-11:
Installed Octoprint and Marlin, the connection via RX TX works.

I installed the klipper on octoprint, the connection via the USB cable is working.

Configured via RX TX, the connection does not work.

in the logs the same connection error.

```
Starting serial connect
Timeout on connect
Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 65, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 249, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 306, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 241, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
```

mks-viva, 2021-04-12:
You can sned 'ls /dev/serial/by-id/*' and get serial port

rom-ka1, 2021-04-13:
[image]

mks-viva, 2021-04-13:
You can try to restart putty.exe

rom-ka1, 2021-04-13:
Restarting. Nothing changed.

rom-ka1, 2021-04-13:
I need to connect via UART, not via a USB cable.

mks-viva, 2021-04-13:
I have not tried using the serial port