makerbase-mks/MKS-SGEN_L-V2 #59: MKS PWC on MSK SGEN L V2 doesn't stay on

ILGROSSOFA, 2021-04-16:
If I press the power button it turns on and after few seconds on display it appears "printers stopped" and it turns off.
If I keep button pressed, it stay on until I release it, then it turns off immediately.

Hardware connections:
PWR_DET connected to pin P1.28
PWR_OFF connected to pin P2.0 servo

My firmware modifications:

**configuration.h**
#define PSU_CONTROL
//#define PSU_NAME "Power Supply"

#if ENABLED(PSU_CONTROL)
  #define PSU_ACTIVE_STATE HIGH      // Set 'LOW' for ATX, 'HIGH' for X-Box

  //#define PSU_DEFAULT_OFF         // Keep power off until enabled directly with M80
  #define PSU_POWERUP_DELAY 250   // (ms) Delay for the PSU to warm up to full power

  #define AUTO_POWER_CONTROL      // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS         // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    #define AUTO_POWER_CONTROLLERFAN
    #define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU over this temperature
    #define POWER_TIMEOUT 30
  #endif
#endif

**Marlin-bugfix\Marlin\src\pins\lpc1769\pins_MKS_SGEN_L_V2.h**
#if ENABLED(PSU_CONTROL)
#ifndef PS_ON_PIN
  #define PS_ON_PIN         P2_00    //PW_OFF, you can change it to other pin
#endif
#ifndef KILL_PIN
  #define KILL_PIN          P1_28    //PW_DET, you can change it to other pin
#endif
#define KILL_PIN_INVERTING  true     //true : HIGH level trigger
#endif

Please help me, thank you!
GROSSO

ILGROSSOFA, 2021-04-16:
I forgot to tell that I have TS35 display.

mks-viva, 2021-04-17:
Can you provide your firmware sourcode to me?

ILGROSSOFA, 2021-04-18:
I solved by myself!!!

This is the correct firmware modification (just add it) of file Marlin\src\pins\lpc1769\pins_MKS_SGEN_L_V2.h:

#if ENABLED(PSU_CONTROL)
#ifndef PS_ON_PIN
  #define PS_ON_PIN         P2_00    //PW_OFF, you can change it to other pin
#endif

The problem was caused by following instructions:
#ifndef KILL_PIN
#define KILL_PIN          P1_28    //PW_DET, you can change it to other pin
#endif
#define KILL_PIN_INVERTING  true     //true : HIGH level trigger

Please rectify instructions at follwing pages for people using MKS SGEN L V2 with TS35 display:
https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/MKS_PWC

Now it works like a charm!

Thank you!
GROSSO