makerbase-mks/MKS-SGEN_L-V2 #61: Error Compiling - MKS SGen_L V2.0

buiduy6868, 2021-04-23:
**Anyone can help one this error, I can not flash to my new board MKS SGen_L V2.0**

In file included from c:\users\khach\appdata\local\temp\arduino_build_566565\sketch\src\inc\MarlinConfigPre.h:48:0,

                 from c:\users\khach\appdata\local\temp\arduino_build_566565\sketch\src\inc\marlinconfig.h:28,

                 from sketch\src\HAL\LPC1768\tft\tft_spi.cpp:23:

c:\users\khach\appdata\local\temp\arduino_build_566565\sketch\src\hal\avr\inc\conditionals_lcd.h:25:4: error: #error "Sorry! TFT displays are not available for HAL/AVR."

   #error "Sorry! TFT displays are not available for HAL/AVR."

    ^~~~~

In file included from c:\users\khach\appdata\local\temp\arduino_build_566565\sketch\src\pins\pins.h:416:0,

                 from c:\users\khach\appdata\local\temp\arduino_build_566565\sketch\src\inc\marlinconfig.h:32,

                 from sketch\src\HAL\LPC1768\tft\tft_spi.cpp:23:

c:\users\khach\appdata\local\temp\arduino_build_566565\sketch\src\pins\lpc1769/pins_MKS_SGEN_L_V2.h:29:4: error: #error "Oops! Make sure you have the LPC1769 environment selected in your IDE."

   #error "Oops! Make sure you have the LPC1769 environment selected in your IDE."

    ^~~~~

In file included from sketch\src\HAL\LPC1768\tft\tft_spi.cpp:27:0:

sketch\src\HAL\LPC1768\tft\tft_spi.h:27:10: fatal error: lpc17xx_ssp.h: No such file or directory

 #include <lpc17xx_ssp.h>

          ^~~~~~~~~~~~~~~

compilation terminated.

exit status 1
Error compiling for board Arduino Mega or Mega 2560.

buiduy6868, 2021-04-23:
I'm using arduino 1.8.12

ILGROSSOFA, 2021-04-23:
I suggest to use Vs code with Platform.io
https://code.visualstudio.com/
https://youtu.be/eq_ygvHF29I

buiduy6868, 2021-04-24:
> I suggest to use Vs code with Platform.io
> https://code.visualstudio.com/
> https://youtu.be/eq_ygvHF29I

THANKS A LOT