makerbase-mks/MKS-SGEN_L-V2 #62: MKS TFT WiFi with ESP3D - possible wrong firmware

NeoMod, 2021-04-28:
I own what should be the latest version of the SGEN L V2, and I tried following the instructions provided in the wiki to have the MKS TFT WIFI MODULE (essentially an ESP8266 on custom PCB) working with ESP3D.

Following the instructions (https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/MKS_TFT_WIFI) I downloaded the two *bin files and flashed them.

The procedure was a success, but after rebooting the module the WiFi Network is different: it appears as an open network called "ESP-F33EB1" instead of the password-protected one mentioned in the guide.

Also, when connecting to this network (which has the IP class in the range of 192.168.4.x) nothing is loaded: the "index.html" filesystem can't be loaded.

I tried compiling and using the latest official version of ESP3D, which works and loads fine on the module, but it can't communicate with the board.

Thanks a lot for any advice you could give.

kacharros, 2022-01-24:
Me ocurre lo mismo.
ESP3D operativo en web pero sin conexión con la impresora, tanto en puerto WIFI como en AUX1.
¿Alguna solución?

NeoMod, 2022-01-25:
I'm closing this issue sine it never received official support, and it seemed to me that this "feature" on that motherboard was more like "smoke and mirrors".

Also, as of today I moved away from MKS products so I can no longer provide insights on this problem. I'm sorry I couldn't be of any help @kacharros: my oonly advice at this point would be to change brand.

closed 2022-01-25 by NeoMod