makerbase-mks/MKS-SGEN_L-V2 #64: MKS FES problem!

ILGROSSOFA, 2021-05-21:
Hi,
I have installed MKS FES and followed instructions indicated here but I have the following problem:

when the filament runout from FES, the trigger works and in fact on the display it appears the info to load filament but after few seconds the printer restart.

After printer restarted no chance to continue the previous printing.

What's wrong with it?

Thanks.

GROSSO

ILGROSSOFA, 2021-05-22:
More infos on my problem:
I tried all display menu options "load filament", "unload filament" and "remove filament" and all of them, after asking me which type of nozzle preheat I need, they restart the printer.

ILGROSSOFA, 2021-05-24:
My problem seems to be a bug of Marlin Bugfix 2.0.x 
I solved by myself installing Marlin 2.0.8.1!
bye
GROSSO

mks-viva, 2021-05-28:
Hi! What screen is you using?

ILGROSSOFA, 2021-06-05:
I use TS35 v2.0 but I solved with new latest Marlin. It was a Marlin bug.