makerbase-mks/MKS-SGEN_L-V2 #67: MKS SGEN_L V2.0 + MKS TMC2209 V2.0 Testing X connection... Error: All LOW

sag3ll0, 2021-06-18:
I have an MKS SGEN_L V2.0 board with MKS TMC2209 V2.0 drivers, the board is configured according to the [instructions](https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/Drivers_MKS_TMC2209) , as the basis, the firmware is used from [here ](https://github.com/makerbase-mks/MKS-SGEN_L-V2/tree/master/Firmware/Marlin-bugfix-2.0.x-MKS%20TS35%2BSGEN_L%20V2/Marlin-bugfix-2.0.x.) The drivers are connected in `uart` mode. 'TMC_DEBUG' mode is enabled. When the code is requested, M122 returns the following result 
```gcode
Send: M122 S1
Recv: axis:pwm_scale/curr_scale/mech_load|flags|warncount
Recv: 		X	Y	Z	E
Recv: Address		0	0	0	0
Recv: Enabled		false	false	false	false
Recv: Set current	800	800	800	800
Recv: RMS current	1436	1436	1436	1436
Recv: MAX current	2025	2025	2025	2025
Recv: Run current	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31
Recv: CS actual	0/31	0/31	0/31	0/31
Recv: PWM scale
Recv: vsense		0=.325	0=.325	0=.325	0=.325
Recv: stealthChop	false	false	false	false
Recv: msteps		256	256	256	256
Recv: tstep		0	0	0	0
Recv: PWM thresh.	1581	1581	1054	843
Recv: [mm/s]		100	100	150	30
Recv: OT prewarn	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false
Recv: pwm scale sum	0	0	0	0
Recv: pwm scale auto	0	0	0	0
Recv: pwm offset auto	0	0	0	0
Recv: pwm grad auto	0	0	0	0
Recv: off time	0	0	0	0
Recv: blank time	16	16	16	16
Recv: hysteresis
Recv:  -end		-3	-3	-3	-3
Recv:  -start		1	1	1	1
Recv: Stallguard thrs	240	240	240	0
Recv: uStep count	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: sg_result	0	0	0	0
Recv: stst		*	*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0x00:00:00:00	 Bad response!
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 		Z	0x00:00:00:00	 Bad response!
Recv: 		E	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... Error: All LOW
```

What is the reason 
`X	0x00:00:00:00	 Bad response!`
`Testing X connection... Error: All LOW`
here configuration of firmware and photo of connected drivers [MKS_issue_data.zip](https://github.com/makerbase-mks/MKS-SGEN_L-V2/files/6676690/MKS_issue_data.zip)

mks-viva, 2021-07-30:
Set TMC uart mode, you need set it
1. Use jumpers connect uart
[image]

2. In Configuration.h file, define driver type is TMC2209
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209

3. You need connect DC power for the board

sag3ll0, 2021-07-30:
> 1. Use jumpers connect uart
[image]

> 2\. In Configuration.h file, define driver type is TMC2209
[image]

> 3\. You need connect DC power for the board external 12v 6a with mks-pwc (not usb)

[image]

all this info in attached MKS_issue_data.zip arch

mks-viva, 2021-07-31:
[image]

sag3ll0, 2021-08-05:
yes, it really works, my mistake is that I looked at the instructions wrong.

closed 2021-08-05 by sag3ll0