makerbase-mks/MKS-SGEN_L-V2 #68: SD Initt Fail w/ REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER not in

bdelia, 2021-07-29:
I am setting up this board for the first time in a TEVO Tornado (Green).  The original MKS Base v1.4 board got shorted and I am trying to upgrade to the MKS SGEN L V2.0.  I am almost good to go for the initial trial, but cannot get the SD card reader to work in the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER (12864) connected to the board.

I have uncommented the following within the Configuration.h file:

```
#define SDSUPPORT
...
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define ST7920_DELAY_1 DELAY_NS(150)
#define ST7920_DELAY_2 DELAY_NS(150)
#define ST7920_DELAY_3 DELAY_NS(150)
```

Any thoughts/help is greatly appreciated.

mks-viva, 2021-07-30:
You need set it:
1. In pins_MKS_SGEN_L_V2.h file （path: Marlin->src->pins->lpc1769）
#define SDCARD_CONNECTION              LCD
2. Complie and copy firmware.bin to tf card, insert motherboard, reset it and update firmware
3. You need remove motherboard's TF and insert SD card to LCD

bdelia, 2021-07-30:
Thank you, I will dive into verifying if that is set up or not later today.

closed 2021-09-30 by mks-viva